lowRISC/ibex #1663: riscv_csr_test fails because we don't model MCAUSE properly

rswarbrick, 2022-05-18:
*Copying from https://github.com/lowRISC/ibex/issues/1658#issuecomment-1129412757 (a silly place for me to have put this text in the first place)*

Running the `riscv_csr_test` with
```
make -j4 ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1 SIMULATOR=xlm WAVES=1
```
fails. It turns out that this is in the self-checking part, where we're writing and then reading back the MCAUSE register. The problem is that we describe the `mcause` CSR to riscv-dv with the following hunk of YAML:
```
# MCAUSE
- csr: mcause
  description: >
    Indicates trap cause
  address: 0x342
  privilege_mode: M
  rv32:
    - field_name: Interrupt
      description: >
        Indicates if trap caused by interrupt
      type: WARL
      reset_val: 0
      msb: 31
      lsb: 31
    - field_name: Exception Code
      type: WLRL
      reset_val: 0
      msb: 4
      lsb: 0
```

This might have been right once, but is no longer correct since Greg implemented internal interrupts. I don't think that `gen_csr_test.py` supports the behaviour that we've got for our MCAUSE field, which boils down to this code in `ibex_cs_registers.sv`:
```systemverilog
      // mcause: exception cause
      CSR_MCAUSE: csr_rdata_int = {mcause_q.irq_ext | mcause_q.irq_int,
                                   mcause_q.irq_int ? {26{1'b1}} : 26'b0,
                                   mcause_q.lower_cause[4:0]};
```

Here, we ostensibly have a 31-bit exception code field, but the top bits are all constrained to be equal.

@GregAC: What do you think we should do? A lame response might be to just use 6 bits and represent internal interrupts starting at 32, without the bit extension. Or we could split the field and just set the top bit? Or we could do some work to generalise the DV code to understand this sort of thing: for example, we could say that the field is 6 bits but the other 25 bits are some sort of mirror. But that means some DV work and (eek) getting a patch into riscv-dv.

My preference is probably to just set the top bit, so it ends up looking like this:
```systemverilog
      // mcause: exception cause
      CSR_MCAUSE: csr_rdata_int = {mcause_q.irq_ext | mcause_q.irq_int,
                                   mcause_q.irq_int,
                                   25'b0,
                                   mcause_q.lower_cause[4:0]};
```

rswarbrick, 2022-05-18:
@hcallahan-lowrisc: FYI, since I know you're triaging these tests too.

GregAC, 2022-07-22:
Sadly this does need a RISC-V DV change. The issue is even with @rswarbrick's suggested change the top two bits have an invalid value, namely `2'b01`. We cannot express a field that allows some but not all values with the current CSR test generator. For this reason `MSTATUS` is also commented out of the CSR file (it has a field taking a privilege mode and for Ibex you can only set it to 2 of 4 possible values).

We could just remove `MCAUSE` from the CSR descriptions also but I think we're better off expanding the RISC-V DV functionality so we can include both `MCAUSE` and `MSTATUS` in this test.

GregAC, 2022-07-27:
See the RISC-V DV PR here: https://github.com/google/riscv-dv/pull/881

This enables us to flexibly specify WARL fields by using a python fragment in the CSR YAML that specifies how values get legalized. Once that is merged we use it here to allow `MCAUSE` and `MSTATUS` in the CSR test.

GregAC, 2022-08-18:
This is completed by #1779 which is undergoing review

PR #1779 (2022-08-18): [dv] Add MCAUSE and MSTATUS to the riscv_csr_test
With the new WARL functionality for the RISCV-DV CSR test generator we
can bring back these CSRs into the test.

Fixes #1663
commit 32801e8d (2022-08-18): [dv] Add MCAUSE and MSTATUS to the riscv_csr_test

With the new WARL functionality for the RISCV-DV CSR test generator we
can bring back these CSRs into the test.

Fixes #1663
closed 2022-08-18 by GregAC