lowRISC/ibex #1669: Spike log issue

shahbazdullu, 2022-05-28:
## Spike log generation

**Spike log file generation issue:**

I am running UVM test-bench for ibex core verification. But during spike log generation, the log file generation does not end despite its size goes mounts to GBs.  Can you please guide me regarding this problem?

**Operating system:**

Ubuntu Linux 18.* 

**Version of the Ibex source code:**

I am using latest version from master branch.

rswarbrick, 2022-05-28:
Thanks for your report. Please could you give us the exact SHA that you're using, together with the command that you ran to see the problem? Many thanks.

shahbazdullu, 2022-05-30:
Thanks for your reply. I am not using any specific SHA. I have simply cloned repo using `git clone https://github.com/lowRISC/ibex.git`. Then in `/dv/uvm/core_ibex` directory, I ran the following command:

`make TEST=riscv_arithmetic_basic_test SEED=1 SIMULATOR=vcs`

[image]

During spike simulation, the spike log size mounts to GBs and doesn't end even though I ran only `10` instruction and one iteration.

Also, can you please suggest me a specific SHA that is stable, so that I may use that.
Thanks!

rswarbrick, 2022-05-30:
@hcallahan-lowrisc: Would you mind taking a look at this? If the answer is just that this will resolve itself when we remove the non-cosim codepath, I'm happy to solve things by doing that.

hcallahan-lowrisc, 2022-05-30:
Hi @shahbazdullu, sorry to hear you are having problems, but thank you for reporting them. Hopefully I can help you out.
I have seen this problem before, of spike running without terminating and producing a giant logfile. We have been doing some refactoring of the build system recently so that may have caused it.

I tried reproducing your issue locally but I see different results. I am using a different simulator but that shouldn't affect spike.

```sh
harrycallahan@harrycallahan-mercury:~/scratch$ git clone https://github.com/lowRISC/ibex.git
Cloning into 'ibex'...
(...)
Resolving deltas: 100% (14854/14854), done.
harrycallahan@harrycallahan-mercury:~/scratch$ cd ibex
harrycallahan@harrycallahan-mercury:~/scratch/ibex$ git checkout origin/master
Note: switching to 'origin/master'.
(...)
HEAD is now at 97a949df [doc] Update coverage plan to point crosses/cp's
harrycallahan@harrycallahan-mercury:~/scratch/ibex$ git rev-parse HEAD
97a949df0283fbae223372f592287f275ae869f7
harrycallahan@harrycallahan-mercury:~/scratch/ibex$ cd dv/uvm/core_ibex
harrycallahan@harrycallahan-mercury:~/scratch/ibex/dv/uvm/core_ibex$ make clean && git clean -fdx
rm -rf out/
harrycallahan@harrycallahan-mercury:~/scratch/ibex/dv/uvm/core_ibex$ make TEST=riscv_arithmetic_basic_test SEED=1 ITERATIONS=1 SIMULATOR=xlm
mkdir -p out/build
Running RTL simulation at out/run/riscv_arithmetic_basic_test.1/rtl.log
Comparing traces for riscv_arithmetic_basic_test.1
Collecting up results (report at out/run/regr.log)
100.00% PASS 1 PASSED, 0 FAILED
```

This checks out the commit 97a949df for me, could you try that commit and run `make clean && git reset --hard HEAD && git clean -fdx` to ensure you have no local changes. (This will delete unstaged/untracked changes)
Many thanks

shahbazdullu, 2022-05-30:
Thanks for your effort. But unfortunately, I am now having issues with `compile-tb.py`.
`
root@lizhen:~/Desktop/ibex_resolve/ibex3/ibex/dv/uvm/core_ibex# make clean && git clean -fdx

rm -rf out/
`
`
root@lizhen:~/Desktop/ibex_resolve/ibex3/ibex/dv/uvm/core_ibex# make TEST=riscv_arithmetic_basic_test SEED=1 ITERATIONS=1 SIMULATOR=vcs
mkdir -p out/build
Traceback (most recent call last):
  File "scripts/compile-tb.py", line 104, in <module>
    sys.exit(main())
  File "scripts/compile-tb.py", line 72, in main
    subprocess.check_output(['pkg-config', '--exists'] + spike_iss_pc)
  File "/usr/lib/python3.6/subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pkg-config': 'pkg-config'
Makefile:331: recipe for target 'out/build/tb/.compile.stamp' failed
make: *** [out/build/tb/.compile.stamp] Error 1
`

hcallahan-lowrisc, 2022-05-30:
Okay, I know this problem. You need to have `pkg-config` installed, and setup environment variables as per [here](https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html?highlight=PKG_CONFIG_PATH#prerequisites-environment-setup). For instance I have locally `PKG_CONFIG_PATH=<my-path-to-spike>/lib/pkgconfig`. This is required as we are moving to a cosimulation flow, and as part of building the testbench we tell the simulator how to link against the spike shared-libraries.
I think our documentation could be a bit better in this area. Hopefully this gets you running! Thanks

rswarbrick, 2022-05-30:
Thanks for looking at this, Harry. Maybe we should catch this error and print out something a bit clearer for people not "in the know" :-) Are you happy to tweak `compile-tb.py` accordingly? Otherwise, I'm happy to take it.