lowRISC/ibex #1671: Fusesoc error while building

tanvir-a, 2022-05-31:
Hi,

I got the following error while building for arty-100 while following the steps as `make build-arty-100`.

```
Traceback (most recent call last):
  File "/home/tanvira/.local/bin/fusesoc", line 8, in <module>
    sys.exit(main())
  File "/home/tanvira/.local/lib/python3.8/site-packages/fusesoc/main.py", line 835, in main
    fusesoc(args)
  File "/home/tanvira/.local/lib/python3.8/site-packages/fusesoc/main.py", line 825, in fusesoc
    args.func(cm, args)
  File "/home/tanvira/.local/lib/python3.8/site-packages/fusesoc/main.py", line 401, in run
    run_backend(
  File "/home/tanvira/.local/lib/python3.8/site-packages/fusesoc/main.py", line 506, in run_backend
    backend.configure([])
  File "/usr/local/lib/python3.8/dist-packages/edalize/edatool.py", line 198, in configure
    self.configure_main()
  File "/usr/local/lib/python3.8/dist-packages/edalize/vivado.py", line 142, in configure_main
    if f["file_type"].startswith("verilogSource"):
KeyError: 'file_type'
make: *** [Makefile:74: build-arty-100] Error 1
```

## My Environment

**EDA tool and version:**
Fusesoc 0.1
Edalize  0.3.3
Python 3.8

**Operating system:**
Ubuntu 20.04.4 LTS

**Version of the Ibex source code:**
pulpissimo-v6.1.1

rswarbrick, 2022-05-31:
Hi there! I think this means you're using an older Edalize than is needed. If you switch to either the special `ot-v0.2` tag or the current tip of master, things should start working.

tanvir-a, 2022-06-01:
Hi @rswarbrick, thanks for your reply. I have updated the `edalize` and it is working.

closed 2022-06-01 by tanvir-a