lowRISC/ibex #1674: Mismatch between Ibex tracelog CSV file and spike log CSV file and test generation problem.

Furqan-Lamp, 2022-06-02:
## Observed Behavior
There were mismatches between the Ibex CSV file and Spike CSV file for a single test. 
The log files generated by the core and spike are same, but when the log files are converted into CSV file, some instructions are skipped which creates a mismatch when comparison is done. Also, i was able to run only 2 tests, the table below shows which ones.

## Expected Behavior
All instructions should appear in the CSV file in their respective order in both files, and the comparison tests should pass. And all tests should work. 

## Steps to reproduce the issue
I was working with post simulation comparison and NOT co-simulation, there was an issue with test generation. Apart from riscv_arithmetic_basic_test and riscv_debug_basic_test, no other tests seem to be working, it gives an error in each one. I made the following table of the tests.

Test : Name | Test Generation | Comparison Test
-- | -- | --
riscv_arithmetic_basic_test | Successfully generates | Fail (Missing Instructions in CSV log)
riscv_machine_mode_rand_test | Error | N/A
riscv_rand_instr_test | Error | N/A
riscv_rand_jump_test | Error | N/A
riscv_jump_stress_test | Error | N/A
riscv_mmu_stress_test | Error | N/A
riscv_illegal_instr_test | Error | N/A
riscv_loop_test | Error | N/A
riscv_hint_instr_test | Error | N/A
riscv_ebreak_test | Error | N/A
riscv_debug_basic_test | Successfully generates | Pass
riscv_debug_stress_test | Error | N/A
riscv_debug_branch_jump_test | Error | N/A
riscv_debug_instr_test | Error | N/A
riscv_debug_wfi_test | Error | N/A
riscv_dret_test | Error | N/A
riscv_debug_ebreakmu_test | Error | N/A
riscv_debug_csr_entry_test | Error | N/A
riscv_irq_in_debug_mode_test | Error | N/A
riscv_debug_in_irq_test | Error | N/A
riscv_single_interrupt_test | Error | N/A
riscv_multiple_interrupt_test | Error | N/A
riscv_nested_interrupt_test | Error | N/A
riscv_interrupt_instr_test | Error | N/A
riscv_interrupt_wfi_test | Error | N/A
riscv_interrupt_csr_test | Error | N/A
riscv_csr_test | Stuck | N/A
riscv_unaligned_load_store_test | Error | N/A
riscv_mem_error_test | Error | N/A
riscv_perf_counter_test | Error | N/A
riscv_debug_single_step_test | Error | N/A
riscv_reset_test | Error | N/A
riscv_rv32im_instr_test | Error | N/A
riscv_user_mode_rand_test | Error | N/A

I ran spike and ibex separately by applying the same seed for riscv_arithmetic_basic_test
command : 
`make iss_run TEST=riscv_arithmetic_basic_test SEED=18 ITERATIONS=1`
`make TEST=riscv_arithmetic_basic_test SEED=18 ITERATIONS=1`

Spike log and trace log files created were the same (spike log and Ibex trace log). There is 2 python scripts which converts the log files to CSV file. Following are the directories :
/ibex/dv/uvm/core_ibex/`ibex_log_to_trace_csv.py.` 
/ibex/vendor/google_riscv-dv/scripts/`spike_log_to_trace_csv`.
The Ibex trace log CSV contained all the instruction, but spike CSV log file skipped some instructions which was not supposed to happen. I checked the log file and the generated CSV file , some instructions which were not added to CSV file causing it to make a mismatch.  

## My Environment
I am running the environment on VCS Synopsys 2016 version.

**Operating system:**
 Ubuntu 18.04

**Version of the Ibex source code:**
 Ibex Source Code  : 
09d2dd23061a12b2eaf68815e1dc1d5488fb805a

GregAC, 2022-06-08:
The log -> CSV process does do some filtering if I remember correctly (the script that does it comes from RISC-V DV). It removes instructions that don't cause register writes like branches or store instructions.

In general I'd recommend avoiding the post simulation comparison flow (for one thing is has some deficiencies, like the filtering). The plan is to support co-simulation only and support for the post simulation comparison will be removed entirely.

Furqan-Lamp, 2022-06-10:
Thank you sir for your reply.  
Okay got it. The reason for using post simulation is that i get a lot of errors when i execute Cosim. And also. When i set num_sub_programs = 0, then google dv generates instructions otherwise it gives me a segmentation fault. The error is as follows.
 
`An unexpected termination has occurred in out/build/instr-gen/vcs_simv due to a signal: Segmentation fault`

Also, I wanted to work on [issue](https://github.com/lowRISC/ibex/issues/1664) . The remaining four. They are related to coverage.

closed 2022-06-14 by Furqan-Lamp