lowRISC/ibex #1676: Verification failed in compiling tb using xlm

YiKangOY, 2022-06-07:
## Observed Behavior

I run `make TEST=riscv_machine_mode_rand_test ITERATIONS=1 SEED=1 COV=0 SIMULATOR=xlm` in `dv/uvm/core_ibex`. The terminal outputs:
`mkdir -p out/build
make: *** [Makefile:331: out/build/tb/.compile.stamp] Error 1`

The contents in compile.log in `out/build/tb` are:
`    export "DPI-C" function simutil_get_scramble_key;
         |
xmvlog: *E,IMPMOD (/home/oyykyook/ibex/rtl/ibex_if_stage.sv,364|9): Export DPI function/task declaration only allowed in module/interface/package/compilation_unit_scope/program_block  declaration.
    export "DPI-C" function simutil_get_scramble_nonce;
         |
xmvlog: *E,IMPMOD (/home/oyykyook/ibex/rtl/ibex_if_stage.sv,365|9): Export DPI function/task declaration only allowed in module/interface/package/compilation_unit_scope/program_block  declaration.
package dv_utils_pkg;
                   |
xmvlog: *W,TSNSPK (/home/oyykyook/ibex/vendor/lowrisc_ip/dv/sv/dv_utils/dv_utils_pkg.sv,5|19): `timescale is not specified for the package.  The default timescale of 1ns/1ns will be assumed for this package.
  endtask: run_cosim_imem_errors;
                                |
xmvlog: *W,UEXPSC (/home/oyykyook/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv,254|32): Ignored unexpected semicolon following SystemVerilog description keyword (run_cosim_imem_errors).
(`include file: /home/oyykyook/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv line 254, file: /home/oyykyook/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_agent_pkg.sv line 19)
        iside_error_queue.pop_front();
                                  |
xmvlog: *W,FUNTSK (/home/oyykyook/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv,112|34): function called as a task without void'().
        riscv_cosim_set_iside_error(cosim_handle, iside_error_queue[0].addr);
                                  |
xmvlog: *W,FUNTSK (/home/oyykyook/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv,118|34): function called as a task without void'().
        iside_error_queue.pop_front();
                                  |
xmvlog: *W,FUNTSK (/home/oyykyook/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv,119|34): function called as a task without void'().
    $value$plusargs("timeout_in_cycles=%0d", timeout_in_cycles);
                  |
xmvlog: *W,NOSYST (/home/oyykyook/ibex/dv/uvm/core_ibex/tests/core_ibex_base_test.sv,83|18): System function '$value$plusargs' invoked as a task. Return value will be ignored.
(`include file: /home/oyykyook/ibex/dv/uvm/core_ibex/tests/core_ibex_base_test.sv line 83, file: /home/oyykyook/ibex/dv/uvm/core_ibex/tests/core_ibex_test_pkg.sv line 32)
      determine_irq_from_txn();
                           |
xmvlog: *W,FUNTSK (/home/oyykyook/ibex/dv/uvm/core_ibex/tests/core_ibex_test_lib.sv,238|27): function called as a task without void'().
      ~id_stage_i.div_sel_ex_o && ~id_stage_i.lsu_req_dec && ~id_stage_i.jump_in_dec);
                                                                                     |
xmvlog: *W,UEXPSC (/home/oyykyook/ibex/dv/uvm/core_ibex/fcov/core_ibex_fcov_if.sv,154|85): Ignored unexpected semicolon following SystemVerilog description keyword (end).
xmvlog: *W,SPDUSD: Include directory /home/oyykyook/ibex/vendor/lowrisc_ip/dv/sv/str_utils given but not used.
xrun: *E,VLGERR: An error occurred during parsing.  Review the log file for errors with the code *E and fix those identified problems to proceed.  Exiting with code (status 1).
` 

## Expected Behavior

## Steps to reproduce the issue

## My Environment

**EDA tool and version:**

XCELIUM21.03.007

**Operating system:**

**Version of the Ibex source code:**

GregAC, 2022-06-08:
Do you have access to Xcelium version '21.09.s006'? That is the version we use to run some internal regressions and it's possible the issues you're seeing here are some version specific bug.

YiKangOY, 2022-06-09:
Thank you, after using Xcelium 21.09, I can run the test, but when I set COV=1, errors occur in `merge-cov.py`

Here is the terminal output:

make TEST=riscv_arithmetic_basic_test SEED=1 ITERATIONS=1 COV=1 SIMULATOR=xlm
Repeating compiling TB because variable COV has changed value.
Running RTL simulation at out/run/riscv_arithmetic_basic_test.1/rtl.log
Comparing traces for riscv_arithmetic_basic_test.1
Collecting up results (report at out/run/regr.log)
100.00% PASS 1 PASSED, 0 FAILED
Thu, 09 Jun 2022 11:51:46 INFO     Creating output directory: out/run/fcov
Thu, 09 Jun 2022 11:51:46 INFO     Processing trace log under out/run
Thu, 09 Jun 2022 11:51:46 INFO     Process ibex log[1/2] : out/run/riscv_arithmetic_basic_test.1/spike_cosim.log.csv
Thu, 09 Jun 2022 11:51:46 INFO     Process ibex log[2/2] : out/run/riscv_arithmetic_basic_test.1/trace_core_00000000.log.csv
Thu, 09 Jun 2022 11:51:46 INFO     Building the coverage collection framework
Thu, 09 Jun 2022 11:51:56 INFO     Collecting functional coverage from 2 trace CSV
Thu, 09 Jun 2022 11:51:56 INFO     Processing batch 1/1
Thu, 09 Jun 2022 11:51:58 INFO     Collecting functional coverage from 2 trace CSV...done
Thu, 09 Jun 2022 11:51:58 INFO     Coverage results are saved to out/run/fcov
Traceback (most recent call last):
  File "/home/oyykyook/ibex/dv/uvm/core_ibex/scripts/merge-cov.py", line 146, in <module>
    sys.exit(main())
  File "/home/oyykyook/ibex/dv/uvm/core_ibex/scripts/merge-cov.py", line 141, in main
    return merge_funs[args.simulator](output_dir, args.verbose, cov_dirs)
  File "/home/oyykyook/ibex/dv/uvm/core_ibex/scripts/merge-cov.py", line 84, in merge_cov_xlm
    merge_ret = run_one(verbose,
  File "/home/oyykyook/ibex/dv/uvm/core_ibex/scripts/scripts_lib.py", line 78, in run_one
    return subprocess.run(cmd,
  File "/home/oyykyook/miniconda3/lib/python3.9/subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/home/oyykyook/miniconda3/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/oyykyook/miniconda3/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'imc'
make: *** [Makefile:429: out/run/coverage/.merge.stamp] Error 1

I checked the `merge_cov_xlm` function in merge-cov.py and I found that "imc" belongs to a cmd, so I guess I did not set the environment variable for Xcelium properly.
Would you please show me how you set the environment variable for Xcelium?
Thanks!

YiKangOY, 2022-06-11:
Hi, I found that I did not install imc, after I installed imc I ran `make TEST=riscv_arithmetic_basic_test ITERATIONS=1 SEED=1 COV=1 SIMULATOR=xlm VERBOSE=1`.
The contents in `fcov` folder are: 
asm_test  compile.history  compile.log  default  ibex_trace_log  seed.yaml  sim_riscv_instr_cov_test_0_0.log  xcelium.d

How can I get the coverage report with the scores in a .txt file?

YiKangOY, 2022-06-11:
Hi, I have installed the imc and I ran `make TEST=riscv_arithmetic_basic_test ITERATIONS=1 SEED=1 COV=1 SIMULATOR=xlm VERBOSE=1
`
But the `cov_report_cg.txt` in the `out/run/coverage/report` folder only have 3 covergroups:
`name                                     CoverGroup Average   CoverGroup Covered
-----------------------------------------------------------------------------------
riscv_instr_pkg                          34.73%               29.90% (3388/11331)
core_ibex_fcov_if                        36.18%               7.80% (207/2653)
core_ibex_pmp_fcov_if.g_pmp_cgs_T        42.25%               20.91% (23/110)
core_ibex_pmp_fcov_if.g_pmp_cgs_T.g_pmp_region_fcov 34.44%               5.23% (16/306)`
How can I get a more detailed coverage report likewise the way I using vcs, in which I can see scores for each coverpoints.

Thank you!

ctopal, 2022-06-23:
Hi,

xlm will produce `index.html` under `<your_core_ibex_dv_path>/out/run/coverage/report` for a report about merged coverage from all tests. The reason behind having another folder named `fcov` there is for gathering the coverage data about the instructions that is auto-generated by `riscv-dv` (So you could think about `fcov` as just another run, like `riscv_arithmetic_basic_test.1`).

Sorry for the delayed response!
Canberk

johngt, 2022-07-19:
I believe that this issue is now best resolved with an improvement / update to the documentation. Closing the issue but referenced in our documentation improvement exercise: [[doc] Documentation Improvement - catch all for documentation issues](https://github.com/lowRISC/ibex/issues/1710)

closed 2022-07-19 by johngt