lowRISC/ibex #1679: Simulating Arty A7 35 example in Vivado causes error due to hack in ibex_if_stage.sv

epsilon537, 2022-06-10:
## Observed Behavior
When I try to simulate the Arty A7-35 example in Vivado, I get an error message in elaborate.log:

ERROR: [XSIM 43-3409] Failed to compile generated C file xsim.dir/top_artya7_behav/obj/xsim_4.c.

Examination of xsim_4.c shows the following problem:

```
int gen_prefetch_buffer.simutil_get_scramble_key(svBitVecVal arg0[SV_PACKED_DATA_NELEMS(128)])
{...

```
That is not a valid function signature.
This error can be traced back to the following snippet in ibex_if_stage,sv:
```
`ifndef SYNTHESIS
    // If we don't instantiate an icache and this is a simulation then we have a problem because the
    // simulator might discard the icache module entirely, including some DPI exports that it
    // implies. This then causes problems for linking against C++ testbench code that expected them.
    // As a slightly ugly hack, let's define the DPI functions here (the real versions are defined
    // in prim_util_get_scramble_params.svh)
    export "DPI-C" function simutil_get_scramble_key;
    export "DPI-C" function simutil_get_scramble_nonce;
    function automatic int simutil_get_scramble_key(output bit [127:0] val);
      return 0;
    endfunction
    function automatic int simutil_get_scramble_nonce(output bit [319:0] nonce);
      return 0;
    endfunction
`endif
```

If I disable this chunk of code, the simulation is working fine.

## Expected Behavior
I would like to be able to simulate top_artya7 using Vivado.

## Steps to reproduce the issue
1. make build-arty-35 program-arty
2. Open generated lowrisc_ibex_top_artya7_0.1.xpr in Vivado
3. Run Simulation

## My Environment
Vivado 2021.2
Ubuntu WSL on Windows 11

**Version of the Ibex source code:**

git rev-parse HEAD 
4ada605b6c5b287feebdb5620066b3f8b6495dc5

I did make some changes for simulation purposes but I verified that I get the same error on the unmodified codebase.

[elaborate.log](https://github.com/lowRISC/ibex/files/8878535/elaborate.log)
[xsim_4.zip](https://github.com/lowRISC/ibex/files/8878542/xsim_4.zip)

epsilon537, 2022-06-12:
I provided the wrong commit above. This is the correct one: 
efd289d

a-will, 2022-06-12:
Note that this appears to be an xsim bug. It has added the generate block's name to the _c_identifier_, and this is incorrect. The standard says _c_identifier_ must default to _function_identifier_ for exported functions, so it must literally be `simutil_get_scramble_key` and `simutil_get_scramble_nonce` in these cases.

ekigwana, 2025-01-16:
We just wasted a day on this issue. We are packaging the ibex as an IP using Vivado v2023.2 and the IP synthesizes and simulates with no issues because we we set
`set_property verilog_define {FPGA_XILINX=1 SYNTHESIS=1 RVFI=0 PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx} [get_filesets sources_1]
set_property verilog_define {FPGA_XILINX=1 SYNTHESIS=1 RVFI=0 PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx} [get_filesets sim_1]`

After integration the next hierarchical layer, simulation on that IP fails when it hits that part of code. Would have been nice if `set_property verilog_define {FPGA_XILINX=1 SYNTHESIS=1 RVFI=0 PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx} [get_filesets ibex_ip]` worked.

Perhaps some help on how to create that code so that the IP can find what it expects. In any case we are not vetting the Ibex beyond executing our simple program correctly.