lowRISC/ibex #1683: [dv] Generic Code Coverage Waiver Tool

GregAC, 2022-06-23:
Waiver files are specific to a particular Ibex RTL revision. We would like a way to specify more generic waivers that can continue to work as line numbers change and code is modified. Investigate a simple tool to do this.

johngt, 2022-06-23:
@GregAC could you point me in the direction of the current waiver file(s) / template and I will see what I can come up with.

GregAC, 2022-06-23:
A waiver file is attached, I believe it matches up with this commit: https://github.com/lowRISC/ibex/tree/0c38b203ddc9a72349f38e364afe1e0f898edce0

Here's an example waiver entry

```xml
    <rule ccType="inst" ccfFlagsMask="35184372089280-42" comment="20" domain="icc" entityName="core_ibex_tb_top/dut/u_ibex_top/u_ibex_core/id_stage_i/decoder_i/264" entityType="block" excTime="1652359681" file="19" im-checksum="1979196359" line="1020" name="exclude" reviewer="unknown" text="{7'b000_0101, 3'b100}: if (RV32B != RV32BNone) alu_operator_o = ALU_MIN;" user="0" vscope="default"></rule>
```

It gives you 'line' and 'text' corresponding to the line that's being waived. It's the 'line' part that would need to change as the file gets modified. The 'entityName' makes things a bit tricky. This is a path in the hierarchy of the design which we cannot easily determine. In particular the final '264' in this example appears to be an index into the structure bottom level module (which is in ibex_decoder.sv)  which I think identifies the precise part of the case statement which is getting waived in this case.

That final number is likely to change with various modifications but there's no easy/obvious way to work out what it should be without using the tool. To start with I think we assume all of the various fields remain constant and it's only really 'line' we're changing. So our tool input will have some kind of regex and filename that will identify a particular line, then the fixed values of all of the other fields in the waiver. It will find the line fill in the line field and output the other fields are provided in the input file.

One possibility is we generated tcl commands that we feed to the tool. This may allow us just to provided filename/line and it will work out the 'entityName' path, but more research is required here. Let's keep our tool flexible so it can produce something that looks like this waiver file and be altered to produce tcl commands instead.

[refine3a.txt](https://github.com/lowRISC/ibex/files/8968112/refine3a.txt)

GregAC, 2022-12-08:
Would be nice to have but moving this out of V3 as it's not essential and a significant amount of work