lowRISC/ibex #1690: Spike (ISS) and Ibex under cosimulation use different Debug Mode Exception Addresses

hcallahan-lowrisc, 2022-06-30:
## Observed Behavior

Under the new cosimulation framework, the following error is commonly encountered : 
`
UVM_FATAL <path>/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv(131) @ 215587: uvm_test_top.env.cosim_agent.scoreboard [ibex_cosim_agent_pkg::ibex_cosim_scoreboard.run_cosim_rvfi] Cosim mismatch Synchronous trap was expected at ISS PC: 808 but DUT didn't report one at PC 80000004
`

This is due to the two systems jumping to different addresses when an exception is encountered while in debug mode. Ibex in its regression testbench uses [(32'h`BOOT_ADDR + 'h4)](https://github.com/lowRISC/ibex/blob/5c49fad9a236351a884f55d44248e4f9a0f7c7e5/dv/uvm/core_ibex/tb/core_ibex_tb_top.sv#L75) as its default, while spike uses 0x808, which is hardcoded in [debug_rom_defines.h](https://github.com/lowRISC/riscv-isa-sim/blob/9af9730baf7b956c3072c1b436d867aca5ef8f4c/riscv/debug_rom_defines.h#L21). This is presumably due to [older versions](https://static.dev.sifive.com/riscv-debug-spec-0.11nov12.pdf) of the riscv-debug-spec which specified 0x808. The pulp debug module also uses 0x808 as its default value [(1)](https://github.com/pulp-platform/riscv-dbg/blob/69be5ddc03ea1688c0eab47d6ed9d0e8725beda1/src/dm_pkg.sv#L30) [(2)](https://github.com/pulp-platform/riscv-dbg/blob/69be5ddc03ea1688c0eab47d6ed9d0e8725beda1/doc/debug-system.md?plain=1#L453).

## Expected Behavior

## Steps to reproduce the issue

## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

johngt, 2022-07-19:
Moved this to a V2 issue. If we are not going to resolve it, let us tag it with V3/MS3 or state that we will not resolve it.

Assigning to @hcallahan-lowrisc as the original reporter until we know if it will be for V2 or later.

GregAC, 2022-07-21:
For V2 we can just alter the fixed address in Spike so it uses the address used by the Ibex DV environment.

Really for full testing we should be moving the boot address around, potentially something to address for V3. There's also an argument we should just use  a fixed debug exception address exactly how spike does that isn't related to the boot address.

GregAC, 2022-07-21:
@hcallahan-lowrisc just switching the fixed spike address should be a quick fix, so let's get that done which in turn will improve our pass rates.

If you make the change, test it and make a PR against our spike fork I can deal with getting that packaged so it's available for CI and such.

hcallahan-lowrisc, 2022-07-23:
@GregAC PR is open now
https://github.com/lowRISC/riscv-isa-sim/pull/8

johngt, 2022-07-28:
@hcallahan-lowrisc  - your PR was merged-in according to link above. Anything else remaining from this issue or safe to close out?

GregAC, 2022-08-10:
I've done a new spike release so this is now integrated into CI

closed 2022-08-10 by GregAC