lowRISC/ibex #1693: [dv] Timeouts across several tests

ctopal, 2022-07-06:
## Observed Behavior

List of timed out tests that are otherwise passing in the daily regression from 04/07/22:
  - riscv_debug_ebreakmu_test (seed 22641)
  - riscv_single_interrupt_test (seed 22641)
  - riscv_nested_interrupt_test (seed 22638)
  - riscv_mem_error_test (seed 22641)
  - riscv_debug_single_step_test (seed 22633)

## Expected Behavior

Since these are passing normally, we should maybe increase the timeout limit to not these have failures.
## Steps to reproduce the issue

Please run the command below to run all the tests listed at once

```make TEST=riscv_debug_ebreakmu_test,riscv_single_interrupt_test,riscv_mem_error_test ITERATIONS=1 SIMULATOR=xlm -j3 SEED=22641; make TEST=riscv_nested_interrupt_test ITERATIONS=1 SIMULATOR=xlm SEED=22638; make TEST=riscv_debug_single_step_test ITERATIONS=1 SIMULATOR=xlm SEED=22633```

GregAC, 2022-07-21:
I'll look at updating the timeout. I can run a test regression on my workstation to quickly determine the impact of a larger timeout.

GregAC, 2022-08-23:
Timeout increased here: https://github.com/lowRISC/ibex/pull/1771

Closing this now as specific timeout causes are being tracked seperately

closed 2022-08-23 by GregAC