lowRISC/ibex #1701: [dv, fcov] Debug Related Coverage Holes

GregAC, 2022-07-13:
The following debug related coverpoints are not yet being hit:

- [ ] Some categories for single stepping over each instruction category - `cp_single_step_instr`
  - [ ] `CSRAccess`
  - [ ] `EBreakDbg`
  - [ ] `EBreakExc`
  - [ ] `ECall`
  - [ ] `MRet`
  - [ ] `MRet`
  - [ ] `DRet`
  - [ ] `WFI`
  - [ ] `Fence`
  - [ ] `FenceI`
  - [ ] `FetchError`
  - [ ] `CompressedIllegal`
  - [ ] `UncompressedIllegal`
  - [ ] `CSRIllegal`
  - [ ] `PrivIllegal`
- [ ] Hardware breakpoint triggered - `cp_insn_trigger_enter_debug`
- [ ] Hardware breakpoint on instruction that causes an exception `cp_insn_trigger_exception`

Note for some of the single step instruction categories it's possible we simply cannot trigger the coverpoint for those. However from an initial look I think it should be possible for all of them (other than `None`)

hcallahan-lowrisc, 2022-07-28:
Update for 2022-07-27 : In progress for `cp_single_step_instr` 2 remaining
                                       Also now 6 remaining for `cp_insn_trigger_enter_debug`

hcallahan-lowrisc, 2022-08-01:
Update for 2022-08-01 : In progress for `cp_single_step_instr`, increased remaining to 4

hcallahan-lowrisc, 2022-08-04:
Update for 2022-08-03 : In progress for `cp_single_step_isntr`, remaining still at 4, revised estimate up +4

hcallahan-lowrisc, 2022-08-11:
Update for 2022-08-10 : In progress for `cp_single_step_instr`, remaining 1

hcallahan-lowrisc, 2022-08-19:
Update for 2022-08-16 : In progress for `cp_insn_trigger_*` coverpoints, remaining 4 for each.
I'm working on a new directed stream for riscv_dv which inserts a short sequence of instructions which enable the trigger and set the address to a randomly-generated insn at the end of the stream. With appropriate randomization this should allow us to hit both coverpoints.

hcallahan-lowrisc, 2022-08-25:
Update for 2022-08-23 : In progress for `cp_insn_trigger_* coverpoints`, remaining 2 for each.

hcallahan-lowrisc, 2022-08-30:
Update for 2022-08-26 : In progress for `cp_insn_trigger_*` coverpoints, remaining 1 for each. Just need to do a little tidyup and create some PR's.

closed 2022-11-01 by GregAC