lowRISC/ibex #1710: [doc] Documentation Improvement - catch all for documentation issues

johngt, 2022-07-19:
Catch all for all issues where users have had a misunderstanding or we have some explicit knowledge that should be shared with the community via OT documentation (e.g. knowing that specific versions work for tools).

johngt, 2022-07-19:
https://github.com/lowRISC/ibex/issues/1669 - log file might grow indefinitely if pkg_config not set up explicitly. Although the issue may be resolved, need to check that in the configuration guide that we capture common errors and highlight specific expected errors if pkg_config is not set, at least for a few issues that have been reported.

johngt, 2022-07-19:
https://github.com/lowRISC/ibex/issues/1646 - I believe that the linked issue can be closed with an update to the documentation

johngt, 2022-07-29:
Not specifically an Ibex DV / documentation issue, but tracking https://github.com/lowRISC/opentitan/issues/2498 as a documentation issue so that we can close that out.

GregAC, 2022-08-31:
Shifting this out of the V2 milestone

GregAC, 2023-02-27:
Closing this as it was only relevant to prior project tracking.

closed 2023-02-27 by GregAC