lowRISC/ibex #1711: [cosim,dv] riscv_mem_error_test failing under IBEX_CONFIG={small,experimental-branch-predictor} (ePMP / Cosim)

johngt, 2022-07-19:
riscv_mem_error_test failing under IBEX_CONFIG={small,experimental-branch-predictor}
see issue - https://github.com/lowRISC/ibex/issues/1656

From Rupert's last point on that issue:
> For the small configuration, at least, the problem seems to be that we're running with no PMP. But the Spike configuration assumes otherwise!
>
> The way to get this to work would be to call processor->set_pmp_num() and processor->set_pmp_granularity() in the  SpikeCosim constructor. Unfortunately, I'm not sure that it's going to be particularly easy to get at the right configuration variables from there: some plumbing required, I suspect.

Unsure if we are still observing the error. If we are not and happy that the related issue is resolved please update that issue as such.

marnovandermaas, 2022-07-21:
@johngt Can we close this issue now since we can add the other issue to the project tracker?

GregAC, 2022-08-10:
Closing as tracked in #1656

closed 2022-08-10 by GregAC