lowRISC/ibex #1718: [dv] Allow generic use of ecall

GregAC, 2022-07-22:
As I understand it RISC-V DV uses `ecall` to the signal the end of a test. The issue with this is we can't have arbitrary `ecall` executions within a generated test.

We should use another mechanism to trigger test end (maybe a write to a specific memory location?) allowing use of `ecall` within the test program body.

I've added this to the V2 milestone as I think we'll need to to close some of the coverage holes (specifically single stepping over an `ecall`).

Depending on how complex it ends up being we could consider waiving the need for this for V2 or do some limited fix that allows us to close the coverage without altering the end of test signalling too radically.

> estimate 8

hcallahan-lowrisc, 2022-07-23:
I think the riscv-dv magic handshake address should be able to handle this. In fact, [their documentation ](https://github.com/google/riscv-dv/blame/master/docs/source/handshake.rst#L114) specifies that the riscv_csr_test does use this mechanism, and I don't really understand why the others don't too. 
Our[ riscv_csr_test uvm code](https://github.com/lowRISC/ibex/blob/a6c182e7be504c20ccd38b5fe5f3a345574aa390/dv/uvm/core_ibex/tests/core_ibex_test_lib.sv#L15) already checks for the TEST_PASS handshake, so we have a working implementation to start from. I would guess it shouldn't be too difficult to adapt it to be the default method for the base test.
Will probably be a very similar fix to what's needed for https://github.com/lowRISC/ibex/issues/1570

johngt, 2022-07-25:
@hcallahan-lowrisc is this something you or @GregAC will be looking at? Will assign to you for now. Tag @johngt if it is for discussion or @GregAC directly if you know it is something he will look at.

GregAC, 2022-07-25:
> I think the riscv-dv magic handshake address should be able to handle this. In fact, [their documentation ](https://github.com/google/riscv-dv/blame/master/docs/source/handshake.rst#L114)specifies [...]

Yes I noticed this after I wrote this issue and did some work on the CSR test. Potentially that'd be a good general mechanism. However it's used for multiple things and is a fairly brittle mechanism. In particular if you wait sitting for a specific write value without understanding what else is going on you may get false triggering.

I think it'd be preferable to switch to another address that's exclusively for indicating test end. Tidying up or outright removing all of the other stuff around the signature address is something else we need to do but it'd be nice to have this without needing that.

hcallahan-lowrisc, 2022-08-01:
@GregAC 
I wonder if it's worth creating some more-flexible API functions to interact with the signature handshaking. 
For instance, the task `check_next_core_status` waits for the next CORE_STATUS type handshake and then performs a DV_CHECK_EQ_FATAL if the payload does not match. Maybe some API functions that don't fail on the first incorrect payload received, but instead would fail only if the expected payload was not received within the timeout. I could also imagine forking this checking process, so as not to block the main thread of the test.
Using a dedicated address for TEST_END/PASS/FAIL seems like a good pragmatic change too. One addr for an in-test general purpose channel, and another for start/stop control flow seems like a good separation of concerns.

hcallahan-lowrisc, 2022-08-08:
Update 2022-08-08 : PR out, was easier than expected. Effort so far = 3. Left 1 remaining for review & feedback.

GregAC, 2022-08-10:
> I wonder if it's worth creating some more-flexible API functions to interact with the signature handshaking.

Yes I think so. I had a long time ago thought we should build some monitor setup that picks up the writes that come to the signature address, determines what they represent, then writes the details to some kind of queue (using a structure that can hold all of the various types of info we can get from signature address writes). Things that currently check what's coming from the signature address would instead look to see if the item they want is in the queue.

The current setup is rather fragile, where a test doesn't see the exact behaviour it expects on the signature address writes it fails, sometimes that may be because of a bug, other times it's because of some perfectly valid thing has happened, just not something the test writer anticipates.

For now I think we can just drop many/all of the checks that rely on the signature address. Though it's a good concept to keep around for belt and braces checking (e.g. test triggers interrupt it can check the exception was actually handled). I think we should look at further splitting up the addresses too. Plenty of address space, no point in making things more complicated for interpreting what's coming in.

PR #1750 (2022-08-10): Change test_done mechanism to use riscvdv handshake, not ecall
Use the address (signature_addr - 0x4) for a TEST_PASS handshake.
Create seperate subscriber port for test_done functionality

By creating a new, distinct port and subscribing to all incoming memory items,
the existing wait_for_mem_txn() can be used with minor modifications to be
able to choose the port to wait on as an argument to the task.

Because the wait_for_mem_txn() implementation currently uses get() to pop the
latest item from the item_collected queue, having two different forked processes
that both await on this queue is  [... cut]
closed 2022-08-10 by hcallahan-lowrisc