lowRISC/ibex #1719: [pmp] M-mode can change locked config in MMWP

marnovandermaas, 2022-07-22:
I'm seeing a divergence in PMP behavior compared to what I believe the spec says.

## Observed Behavior

When MMWP (machine mode whitelist policy) is enabled and RLB (rule lock bypass) is disabled, ibex still allows changing config registers. See this RVFI trace extract:
```
19527	       874	80000278	30f80813	addi	x16,x16,783	 x16:0x58433000 x16=0x5843330f
19947	       895	8000027c	3a081873	csrrw	x16,pmpcfg0,x16	 x16:0x5843330f x16=0x17978d0c
19967	       896	80000280	3a081873	csrrw	x16,pmpcfg0,x16	 x16:0x17978d0c x16=0x18978d0f
```

## Expected Behavior

Expected behavior is what Spike does and disallows the increasing of privilege of the PMP region 0:
```
core   0: 0x80000278 (0x30f80813) addi    a6, a6, 783
core   0: 3 0x80000278 (0x30f80813) x16 0x5843330f
core   0: 0x8000027c (0x3a081873) csrrw   a6, pmpcfg0, a6
core   0: 3 0x8000027c (0x3a081873) x16 0x17978d0c c928_pmpcfg0 0x18978d0c
core   0: 0x80000280 (0x3a081873) csrrw   a6, pmpcfg0, a6
core   0: 3 0x80000280 (0x3a081873) x16 0x18978d0c c928_pmpcfg0 0x17978d0c
```

## Steps to reproduce the issue

Using the following commands from the root of the ibex repo:
```console
cd dv/uvm/core_ibex
make -j7 --keep-going IBEX_CONFIG=opentitan ITERATIONS=1 SEED=2628 TEST=riscv_pmp_full_random_test WAVES=0 COV=0
```

## My Environment

**EDA tool and version:**

Using Spike version ibex_cosim on commit: 9af9730baf7b956c3072c1b436d867aca5ef8f4c

**Operating system:**

Ubuntu Linux 20.04

**Version of the Ibex source code:**

Currently running inside this PR: https://github.com/lowRISC/ibex/pull/1695

ctopal, 2022-07-25:
I am not fully knowledgable around this subject but as far as I understand from the spec, MMWP is strictly related with how Machine Mode behaves regarding memory accesses. Since editing a CSR (`pmpcfg0` in this case) is not technically a memory access, as long as `locked` field of that CSR is clear, we would expect that CSR to change. @GregAC, @vogelpi am I missing something?

ctopal, 2022-07-27:
By the way I also see the same behaviour with ` MSECCFG: MML 0, MMWP 0, RLB 0`. Probably something to do with how spike handles locked fields of pmpcfgs. I am currently trying to figure this out.

ctopal, 2022-07-28:
I think this was a spike bug as I suspected, I might still be wrong. Please check https://github.com/lowRISC/riscv-isa-sim/pull/10 for more detail. Although now, the seed you provided fails with the error described in https://github.com/lowRISC/ibex/issues/1720

marnovandermaas, 2022-08-03:
Good catch, the TOR lock bit should only apply to the address of the previous entry not the config.

closed 2022-08-09 by marnovandermaas