lowRISC/ibex #1720: [dv] Exception divergence using MML PMP mode

marnovandermaas, 2022-07-22:
I'm seeing a divergence between Spike and Ibex when in MML mode and setting region 0 to read only. Spike throws a trap and Ibex zeroes out a destination register. I'm not sure if this is an issue with Spike, Ibex or with the UVM comparison.

## Observed Behavior

Cosimulation result:
```
name:            riscv_epmp_mml_read_only_test
seed:            14822
binary:          out/run/riscv_epmp_mml_read_only_test.14822/test.o
uvm_log:         out/run/riscv_epmp_mml_read_only_test.14822/rtl.log
rtl_trace:       out/run/riscv_epmp_mml_read_only_test.14822/trace_core_00000000.log
rtl_trace_csv:   out/run/riscv_epmp_mml_read_only_test.14822/trace_core_00000000.log.csv
iss_trace:       
iss_trace_csv:   
en_cosim:        True
cosim_trace:     
cosim_trace_csv: 
comparison_log:  
passed:          False
failure_message: |-
  UVM_FATAL /home/mvdmaas/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv(131) @ 7304: uvm_test_top.env.cosim_agent.scoreboard [ibex_cosim_agent_pkg::ibex_cosim_scoreboard.run_cosim_rvfi] Cosim mismatch Synchronous trap was expected at ISS PC: 80001900 but DUT didn't report one at PC 80000118
  [FAILED]: sim error seen
```

Ibex RVFI trace:
```
Time	Cycle	PC	Insn	Decoded instruction	Register and memory contents
           2524	        24	80000080	400015b7	lui	x11,0x40001	 x11=0x40001000
           2864	        41	80000084	10658593	addi	x11,x11,262	 x11:0x40001000 x11=0x40001106
           3004	        48	80000088	30159073	csrrw	x0,misa,x11	 x11:0x40001106  x0=0x00000000
           3024	        49	8000008c	0002c397	auipc	x7,0x2c	  x7=0x8002c08c
           3444	        70	80000090	c5838393	addi	x7,x7,-936	  x7:0x8002c08c  x7=0x8002bce4
           3724	        84	80000094	00002597	auipc	x11,0x2	 x11=0x80002094
           3744	        85	80000098	86c58593	addi	x11,x11,-1940	 x11:0x80002094 x11=0x80001900
           3944	        95	8000009c	0015e593	ori	x11,x11,1	 x11:0x80001900 x11=0x80001901
           4364	       116	800000a0	30559073	csrrw	x0,mtvec,x11	 x11:0x80001901  x0=0x00000000
           4384	       117	800000a4	00002b17	auipc	x22,0x2	 x22=0x800020a4
           4404	       118	800000a8	2f0b0b13	addi	x22,x22,752	 x22:0x800020a4 x22=0x80002394
           4424	       119	800000ac	002b5b13	srli	x22,x22,0x2	 x22:0x80002394 x22=0x200008e5
           4584	       127	800000b0	3b0b1073	csrrw	x0,pmpaddr0,x22	 x22:0x200008e5  x0=0x00000000
           4604	       128	800000b4	74725073	csrrwi	x0,0x747,4	  x0=0x00000000
           4924	       144	800000b8	08c00b13	addi	x22,x0,140	  x0:0x00000000 x22=0x0000008c
           4944	       145	800000bc	3a0b1073	csrrw	x0,pmpcfg0,x22	 x22:0x0000008c  x0=0x00000000
           4964	       146	800000c0	7472d073	csrrwi	x0,0x747,5	  x0=0x00000000
           4984	       147	800000c4	00002b17	auipc	x22,0x2	 x22=0x800020c4
           5024	       149	800000c8	2d0b0b13	addi	x22,x22,720	 x22:0x800020c4 x22=0x80002394
           5124	       154	800000cc	6589	c.lui	x11,0x2	 x11=0x00002000
           5544	       175	800000ce	66658593	addi	x11,x11,1638	 x11:0x00002000 x11=0x00002666
           5564	       176	800000d2	9b2e	c.add	x22,x11	 x22:0x80002394 x11:0x00002666 x22=0x800049fa
           5584	       177	800000d4	002b5b13	srli	x22,x22,0x2	 x22:0x800049fa x22=0x2000127e
           5604	       178	800000d8	3b1b1073	csrrw	x0,pmpaddr1,x22	 x22:0x2000127e  x0=0x00000000
           5664	       181	800000dc	00002b17	auipc	x22,0x2	 x22=0x800020dc
           5804	       188	800000e0	2b8b0b13	addi	x22,x22,696	 x22:0x800020dc x22=0x80002394
           5824	       189	800000e4	6595	c.lui	x11,0x5	 x11=0x00005000
           5944	       195	800000e6	ccc58593	addi	x11,x11,-820	 x11:0x00005000 x11=0x00004ccc
           5964	       196	800000ea	9b2e	c.add	x22,x11	 x22:0x80002394 x11:0x00004ccc x22=0x80007060
           6364	       216	800000ec	002b5b13	srli	x22,x22,0x2	 x22:0x80007060 x22=0x20001c18
           6384	       217	800000f0	3b2b1073	csrrw	x0,pmpaddr2,x22	 x22:0x20001c18  x0=0x00000000
           6644	       230	800000f4	00002b17	auipc	x22,0x2	 x22=0x800020f4
           6704	       233	800000f8	2a0b0b13	addi	x22,x22,672	 x22:0x800020f4 x22=0x80002394
           6724	       234	800000fc	659d	c.lui	x11,0x7	 x11=0x00007000
           6744	       235	800000fe	33258593	addi	x11,x11,818	 x11:0x00007000 x11=0x00007332
           6764	       236	80000102	9b2e	c.add	x22,x11	 x22:0x80002394 x11:0x00007332 x22=0x800096c6
           6784	       237	80000104	002b5b13	srli	x22,x22,0x2	 x22:0x800096c6 x22=0x200025b1
           6924	       244	80000108	3b3b1073	csrrw	x0,pmpaddr3,x22	 x22:0x200025b1  x0=0x00000000
           7104	       253	8000010c	8f8f9b37	lui	x22,0x8f8f9	 x22=0x8f8f9000
           7124	       254	80000110	f8fb0b13	addi	x22,x22,-113	 x22:0x8f8f9000 x22=0x8f8f8f8f
           7264	       261	80000114	3a0b1073	csrrw	x0,pmpcfg0,x22	 x22:0x8f8f8f8f  x0=0x00000000
           7284	       262	80000118	00002b17	auipc	x22,0x2	 x22=0x80002118
           7304	       263	8000011c	27cb0b13	addi	x0,x0,636	  x0:0x00000000  x0=0x00000000
[... 90 lines cut ...]
## Expected Behavior

Not sure if it's a bug in Spike, Ibex or the cosimulation environment

## Steps to reproduce the issue

Using the following commands from the root of the ibex repo:
```console
cd dv/uvm/core_ibex
make -j7 --keep-going IBEX_CONFIG=opentitan ITERATIONS=1 SEED=14822 TEST=riscv_epmp_mml_read_only_test WAVES=0 COV=0
```

## My Environment

**EDA tool and version:**

Using Spike version ibex_cosim on commit: 9af9730baf7b956c3072c1b436d867aca5ef8f4c

**Operating system:**

Ubuntu Linux 20.04

**Version of the Ibex source code:**

Currently running inside this PR: https://github.com/lowRISC/ibex/pull/1695

ctopal, 2022-07-29:
Okay I think I understood what's going on. The first problem is about how we let RVFI know that there is a problem with the issued instruction. However, when I fixed that locally I also came across another problem in which we didn't let ID/EX stage know that the current instruction of it has a PMP error -we actually set the flag but it's late. However, even if I do those two fixes we will be stuck with the problem of having infinite loops coming from the nature of how we configure PMP regions. Same infinite loop happens in `pmp_full_random_test` as well.

marnovandermaas, 2022-10-13:
Closing this because the remaining issue of infinite loops is captured in this issue: https://github.com/lowRISC/ibex/issues/1823

closed 2022-10-13 by marnovandermaas