lowRISC/ibex #1731: [dv] Add ecc error generation to memory agent

GregAC, 2022-07-28:
We need to stimulate ECC errors. The incoming bus data has integrity bits. The driver should provide incorrect versions of these at configurable random internals in a similar manner to how bus errors work.

> estimate 8

GregAC, 2022-07-28:
@ctopal decided to split this into two issues after all, see cosim issue here: https://github.com/lowRISC/ibex/issues/1732

I've left that unassigned for now. Potentially it's something I can look at as I'm more familiar with what's needed. Work on the agent side of things for now. I'll let you know if I find time for the cosim stuff and if not you can get going on that when you need it.

ctopal, 2022-09-20:
>remaining 2022-09-20 3

GregAC, 2022-11-01:
I have a set of commits on top of Canberk's work in https://github.com/lowRISC/ibex/pull/1811 but no PR yet. Once that is merged this will be complete.

GregAC, 2022-11-16:
Completed by https://github.com/lowRISC/ibex/pull/1907

closed 2022-11-16 by GregAC