lowRISC/ibex #1738: Ibex does not reflect data changes to ICache

ctopal, 2022-08-01:
If ICache is enabled and a write to a cached instruction memory (which we are allowed to do) occurred, Ibex does not let ICache know such a store happened. As a result, cached instruction will not change inside ICache (but it will change in the memory component) and when PC will point to that address, Ibex will use the cached instruction. That causes mismatches between ISS and RTL.

It is probably not the most important corner case but I thought it might be good to document it as an issue for future.

hcallahan-lowrisc, 2022-08-01:
This sounds like one of the [2 hard problems in computer science.](https://twitter.com/codinghorror/status/506010907021828096?lang=en)

GregAC, 2022-08-25:
It's acceptable for our icache to not be coherent so I'd wouldn't class this as an RTL bug, more a documentation issue.

Though we probably shouldn't be overwriting code in RISCV-DV generated programs.