lowRISC/ibex #1739: [dv,pmp] Timeouts because of infinite loops in the generated program

ctopal, 2022-08-01:
@marnovandermaas and I talked about this internally so I'm assigning him because of that. Please feel free to unassign yourself if you feel like it :)

In the case of a random PMP configuration in which we don't allow execution at the address of exception handler, we end up going inside an infinite loop (which is totally legal). We need to flag those cases as passing because we want their coverage data in our regression results as they cover the cases where `pmpcfg0` is configured to random values. This issue tracks that task.

marnovandermaas, 2022-08-02:
If I remember correctly, we talked about the case when the instruction fault trap handler is set to non-executable. We can catch this case by checking whether two RVFI packets in a row are exactly the same with the trap flag set. With the trap flag not set, it could be a spinning loop, which we might want to allow in certain cases.

marnovandermaas, 2022-08-04:
We discussed this in the meeting and Greg mentioned that we could check the signal Ibex produces on a nested double trap (a trap raised within the trap handler). In UVM, we should be able to check whether Spike also sees a nested double trap. When we see a certain number (say 1,024) nested double traps in a test, we can decide to terminate the test. For the full random PMP test, we should have a flag that says that this termination of the test should return success while by default it would end up as a failure.

closed 2022-10-06 by GregAC