lowRISC/ibex #1741: [dv] Add coverage for extra ePMP behaviours

GregAC, 2022-08-02:
We need coverpoints documentating and added for the following:

* [ ] Attempting to execute code that belongs to no suitable region in M mode (i.e. one that allows execution from M mode) when `mseccfg.MML` is set.

  - Cover cases where there is a region that matches but no X privilege provided and cases where no regions matches.
* [ ] Attempting to alter a `pmpcfg` CSR to add executable privileges to an M mode or shared region when `mseccfg.MML` is set
  - Cover with both `mseccfg.RLB` set and unset

> estimate 4

ctopal, 2022-08-08:
@GregAC Do you mind if I take this?

GregAC, 2022-08-09:
@ctopal sure

ctopal, 2022-10-24:
I am looking at our coverage implementation and it looks like for the first item, we already define some bins for it, although they are kind of buried deep. 

Would you prefer adding some redundant crosses/coverpoints with this specific conditions or adding documentation to point out in which cross we can see this condition holding true?

For example `pmp_iside_priv_bits_cross` has a autogenerated bin like this one: 

[image]

Which basically is a `cover case where there is a region that matches but no X privilege provided` with `mseccfg.mml` enabled.

And we have `pmp_iside_nomatch_cross` for checking for  `mseccfg.mml` enabled and no-match case for U-mode and M-mode.

ctopal, 2022-10-24:
For the second item we have a more general `cp_edit_locked_pmpcfg` for each pmpcfg register. I believe I can turn that into a cross that has two bins: writes to locked pmpcfg that changes execute access + anything else. @GregAC Would that be enough for this task?

GregAC, 2022-11-01:
> @GregAC Would that be enough for this task?

Yes I think so, just need to make sure to cross in `MML` and `RLB` settings

GregAC, 2022-11-18:
Implemented by https://github.com/lowRISC/ibex/pull/1947

closed 2022-11-18 by GregAC