lowRISC/ibex #1742: Generating large NAPOT regions

marnovandermaas, 2022-08-02:
At the moment RISCV-DV generates addresses randomly including when the PMP encoding is set to NAPOT. Since the NAPOT range is encoded in the address, it means that it is unlikely that we will generate larger regions, since these require a series of ones at the end of the address (see Table 3.11 in the [RISC-V privileged spec](https://github.com/riscv/riscv-isa-manual/releases/download/Priv-v1.12/riscv-privileged-20211203.pdf)).

We should constrain RISCV-DV full randomization to first randomly select a random range and then randomly generate the top bits of the address to create more coverage.

> estimate 8

GregAC, 2022-08-03:
Add a sentence about adding coverage for this, shouldn't be complex given the rest of the PMP stuff is already there so didn't alter the estimate.

marnovandermaas, 2022-09-28:
I made a first attempt to adding coverage points for large NAPOT regions: https://github.com/lowRISC/ibex/pull/1818
In the process I removed some impossible to reach PMP coverage: https://github.com/lowRISC/ibex/pull/1817
> spent 3

marnovandermaas, 2022-10-03:
It took me a while to find out that the Spike version I was using didn't compile because `xrun` gave a vague error.

Besides that I updated the PR https://github.com/lowRISC/ibex/pull/1818 based on suggestions from @ctopal.

> spent 4

marnovandermaas, 2022-10-13:
There is a PR for RISCV-DV, which generates these large NAPOT regions now: https://github.com/google/riscv-dv/pull/903

GregAC, 2022-10-25:
Cover points still need to be added for this.

closed 2022-10-25 by GregAC