lowRISC/ibex #1753: [dv] Testbench memory region checker

GregAC, 2022-08-10:
In https://github.com/lowRISC/ibex/pull/1745#issuecomment-1209244008 @hcallahan-lowrisc highlights that certain memory acceses generated by a particular test may be incorrect in the sense that we don't intend for the test to access that area of memory, but there's some bug in the generated code we're running.

To help catch these create a configurable access checker that connect to a memory bus monitor. It should be configurable to be able to produce warnings or error when certain accesses are seen to certain regions.

Defining exact capabilities (e.g. more precise definitions of 'certain' above) forms part of this task.

hcallahan-lowrisc, 2022-08-10:
A checker in the form of a bus monitor sounds like a much better way to solve this problem. Good idea!

hcallahan-lowrisc, 2022-08-10:
A feature like this could definitely be helpful in designing effective PMP tests, it is currently easy to randomly create tests that advance through a very large memory region one pmp access violation at a time, taking forever and providing little value in coverage. 
Probably more uses that I can't think of right now too.