lowRISC/ibex #1758: [dv] Verify fetch enable

GregAC, 2022-08-11:
Verify the fetch enable signal. Write a sequence that can toggle it for the new sequence library. Use this sequence to randomly toggle fetch enable in some tests. It should have no impact on architectural execution (i.e. timing of instructions will differ but co-simulation checking should not find any issues).

Add a new checker that ensures that when fetch is disabled Ibex isn't fetching and executing new instructions (assertions may be sufficient)

> estimate 4

ctopal, 2022-10-05:
@GregAC Please go ahead I'm still working on the debug side unfortunately..

GregAC, 2022-10-25:
Fetch enable testing added via this PR https://github.com/lowRISC/ibex/pull/1844

closed 2022-10-25 by GregAC