lowRISC/ibex #1759: [dv] Verify ICache Integrity and Scrambling

GregAC, 2022-08-11:
Verify the ICache integrity and scrambling features

For integrity create a testbench component to corrupt icache data and icache tags. Ensure corruptions triggered the required alert. This will require targeting to corrupt data and tags that relate to instructions that are going to get execute. Understanding of scrambling will need to be employed to determine the appropriate memory locations to corrupt.

For scrambling I'm uncertain if we need to do anything further as we're using a existing OT primitive component. @ctopal any thoughts from your OTBN experience? 

> estimate 16

ctopal, 2022-08-11:
In OTBN, this checking of responses to integrity failures is done as this:

1. Read scrambled data from RAM using `encrypt_sram_addr` and `decrypt_sram_data` from `sram_scrambler_pkg`.
2. Apply 1 or 2 bit flips to generate an integrity error when said data gets read back.
3. Write corrupted data to scrambling memory using `encrypt_sram_addr` and `encrypt_sram_data` from `sram_scrambler_pkg`.

A task will need to include incorporating  `sram_scrambler_pkg` to our core_ibex DV environment (or maybe block level ICache DV environment?) and writing up the sequence that uses the functionality mentioned above.

For verifying scrambling itself, OTBN side maps smoke and calls it a day (not sure if it's good to do that, maybe we need to have a talk about it as well?) https://github.com/lowRISC/opentitan/pull/13335#discussion_r903922037

andreaskurth, 2022-10-14:
I've looked into the integrity verification, and I think the following approach should work:
- For each data and tag memory, track which memory entries have been written thus contain valid values (using the `req_i`, `write_i`, and `addr_i` signals).
- For each data and tag memory, track which memory entries are being read (using the same three signals). If an entry is being read that has been written before, corrupt one bit in the output data with a certain probability.
- If the corruption happened, check that the required alert is fired.

I'm estimating 4 for that.

GregAC, 2022-10-14:
Thanks @andreaskurth I will update the estimate with your new one and assign you to the issue.

andreaskurth, 2022-11-04:
Closing this as completed in #1879.

closed 2022-11-04 by andreaskurth