lowRISC/ibex #1760: [dv] Verify double fault behaviour

GregAC, 2022-08-11:
Ibex has a double fault detector. A checker should be created to ensure this functions correctly and double faults should be stimulated in some tests.

No estimate yet as this may be covered by other ongoing work.

GregAC, 2022-09-09:
Added an estimate to this. The cosim side of things is handled by https://github.com/lowRISC/ibex/pull/1808 what remains is

- Produce a double fault in some test (either a new dedicated one or add them to some existing test that already generates exceptions). A read of `cpuctrlsts` that will see the `double_fault_seen` bit being set should be included. The ideal test would clear this bit, recover from the double fault and continue (some of time, other times some loop of faulting that times out would be good).
- A checker for the `double_fault_seen_o` output. This could be done as an assertion along with some tracking logic.

johngt, 2022-10-19:
Taking @marnovandermaas off this task assignment for now.

GregAC, 2022-10-25:
@GregAC to provide more detailed guidance on checking here

GregAC, 2022-11-01:
Reassigning to @GregAC I should be able to address this rapidly freeing @hcallahan-lowrisc for other tasks

GregAC, 2022-11-08:
PR https://github.com/lowRISC/ibex/pull/1916/files to add the double fault checker

GregAC, 2022-11-16:
Closed by https://github.com/lowRISC/ibex/pull/1916

closed 2022-11-16 by GregAC