lowRISC/ibex #1761: [dv] Add instructions to enable/disable random instruction generation to existing tests

GregAC, 2022-08-11:
Ensure CSR writes that will enable and disable the random instruction generator are emitted for some tests. This is mostly dealt with via on-going CSR work. The task here is just to use the new functionality to stimulate the random instruction generator. Some extra work may be required to ensure the randomly generated instructions don't get fed into co-sim checking (but we do ensure they have no effect on architectural state).

> estimate 2

GregAC, 2022-09-09:
#1808 gives us the basis of this (random `CPUCTRLSTS` writes will enable/disable random instruction generation).

Currently the dummy instructions are simply not fed through the RVFI interface so the cosim is unaware of their existence. We should add some extra checking to ensure they don't write to the register file (e.g. some assertion that states that if writeback writes to the RF then the first RVFI stage must be valid, as for dummy instructions we prevent it from entering the RVFI stages).

andreaskurth, 2022-10-14:
According to @GregAC, what basically remains to be done here is adding the assertions that check dummy instructions don't cause register writes. #1853 deals with this.

andreaskurth, 2022-10-25:
Closing as complete with #1853 merged.

closed 2022-10-25 by andreaskurth