lowRISC/ibex #1770: Report a bug in Ibex

hcallahan-lowrisc, 2022-08-12:
## Observed Behavior

Upon executing a WFI instruction in this test, Spike triggers an illegal instruction exception, while Ibex does not. 
The generates a cosimulation mismatch.
In the waveform (extract shown below), the Ibex can be seen going to it's SLEEP mode.

Not exactly sure why this is the case, but it could be due to:
- Inconsistency between Spike and Ibex around MCAUSE.TW (Timeout-Wait) behaviour. This may just be due to the way we configure spike as part of the verification environment. Note that MCAUSE.TW = 0 in the ibex during the failing behaviour (see waveform).  
Relevant Spec Extract [Privileged Architecture Document Version 20211203 p.25] : 
> The TW (Timeout Wait) bit is a WARL field that supports intercepting the WFI instruction (see
> Section 3.3.3). When TW=0, the WFI instruction may execute in lower privilege modes when
> not prevented for some other reason. When TW=1, then if WFI is executed in any less-privileged
> mode, and it does not complete within an implementation-specific, bounded time limit, the WFI
> instruction causes an **illegal instruction exception**. The time limit may always be 0, in which case
> WFI always causes an illegal instruction exception in less-privileged modes when TW=1.

Log Extract
> ```
> riscv_debug_single_step_test.2306
> ---------------------------------
> binary:          test.bin
> rtl_log:         rtl_sim.log
> rtl_trace:       trace_core_00000000.log
> iss_cosim_trace: spike_cosim_trace_core_00000000.log
> 
> [FAILURE]: sim error seen in 'rtl_sim.log'
> ---------------*LOG-EXTRACT*----------------
>     69: SDI/Verilog Transaction Recording Facility Version 21.09-s006
>     70: UVM_INFO /home/harry/projects/ibex/dv/uvm/core_ibex/tests/core_ibex_base_test.sv(174) @ 2020: uvm_test_top [uvm_test_top] Running test : /home/harry/projects/ibex/dv/uvm/core_ibex/out/run/riscv_debug_single_step_test.2306/test.bin
>     71: UVM_INFO /home/harry/projects/ibex/dv/uvm/core_ibex/common/ibex_mem_intf_agent/ibex_mem_intf_response_seq_lib.sv(26) @ 2020: uvm_test_top.env.instr_if_response_agent.sequencer@@instr_intf_seq [uvm_test_top.env.instr_if_response_agent.sequencer.instr_intf_seq] is_dmem_seq: 0x0
>     72: UVM_INFO /home/harry/projects/ibex/dv/uvm/core_ibex/common/ibex_mem_intf_agent/ibex_mem_intf_response_seq_lib.sv(26) @ 2020: uvm_test_top.env.data_if_response_agent.sequencer@@data_intf_seq [uvm_test_top.env.data_if_response_agent.sequencer.data_intf_seq] is_dmem_seq: 0x1
>     73: core_ibex_tb_top.dut.u_ibex_tracer.printbuffer_dumpline.unmblk1: Writing execution trace to /home/harry/projects/ibex/dv/uvm/core_ibex/out/run/riscv_debug_single_step_test.2306/trace_core_00000000.log
> [E] 74: UVM_FATAL /home/harry/projects/ibex/dv/uvm/core_ibex/common/ibex_cosim_agent/ibex_cosim_scoreboard.sv(138) @ 33120: uvm_test_top.env.cosim_agent.scoreboard [uvm_test_top.env.cosim_agent.scoreboard] Cosim mismatch Synchronous trap was expected at ISS PC: 80002600 but the DUT didn't report one at PC 80003468
>     75: 
>     76: 
>     77: --- RISC-V UVM TEST FAILED ---
>     78: 
> --------------------------------------------
> ```

Ibex Trace Extract
[image]

Spike Trace Extract
[image]

[image]

## Expected Behavior

## Steps to reproduce the issue

REPRO
```
cd dv/uvm/core_ibex/
make TEST=riscv_debug_single_step_test ITERATIONS=1 SEED=2306 -j12 -ik WAVES=1
```

> estimate 8

ctopal, 2022-08-16:
I thought https://github.com/lowRISC/riscv-isa-sim/pull/9 would have fixed this problem. I'll check it further, thanks for opening this @hcallahan-lowrisc

ctopal, 2022-08-16:
I ran the command you provided above with ibex-cosim-v0.4 tagged spike, test seems to be passing. I'll try to see the case by running more seeds.

closed 2022-08-16 by ctopal