lowRISC/ibex #1772: Cosim mismatch when reading DCSR.cause

hcallahan-lowrisc, 2022-08-15:
## Observed Behavior
Test failures with the following error : 
```
UVM_FATAL ibex_cosim_scoreboard.sv(138) @ 4049807: Cosim mismatch Register write data mismatch to x22 DUT: 40000100 expected: 400000c0
```

This fails upon reading from `DCSR.cause`
In this case : 

[image]
(IBEX)

[image]
(SPIKE)

Observing the waveform below, we can see that `debug_req_i` is de-asserted shortly after `debug_mode_i` goes low due to leaving debug mode from a dret. We do re-enter debug mode, but the cause can be seen to be recorded as `DBG_CAUSE_STEP`. In spike, we read back `DBG_CAUSE_HALTREQ` at the same instruction, hence the mismatch.

## Expected Behavior

Ibex should record `DCSR.cause` as DBG_CAUSE_HALTREQ.

I think this might be due to the [ibex_controller.sv ](https://github.com/lowRISC/ibex/blob/2f8dfa9dfe2d938228118ed8ab6a543804d0c9ae/rtl/ibex_controller.sv#L662)state machine sampling the `debug_req_i` signal to determine the correct `cause` one cycle later than the decision to enter debug_mode.

## Steps to reproduce the issue
```
cd dv/uvm/core_ibex/
make TEST=riscv_debug_basic_test ITERATIONS=1 WAVES=1 SEED=9892 -j12 -ik
```
[image]

> estimate 8

ctopal, 2022-09-27:
> remaining 2022-09-27 4

hcallahan-lowrisc, 2022-10-13:
> remaining 2022-10-12 1

PR #1816 (2022-10-14): [rtl] Change how we record debug causes
This commit changes when we catch the debug causes. Since `debug_cause_o` only gets saved when `csr_save_cause_o` is high, it would work if we change the cause with a mux that is connected to the input signals.

Resolves #1772
commit f2c1d6dc (2022-10-14): [rtl] Change how we record debug causes

This commit changes when we cath the debug causes. Since debug_cause_o
only gets latched when `csr_save_cause_o` is high, it would work if
we change the cause with a mux that is connected to the input signals.

Resolves #1772
closed 2022-10-14 by hcallahan-lowrisc