lowRISC/ibex #1774: Report a bug in Ibex

marnovandermaas, 2022-08-15:
## Observed Behavior

Divergence between Spike and Ibex when writing to `pmpcfg0`. With an initial value of `0x8B00_0D05` and writing `0x676F_290F`, there are two different behaviors:
- Ibex: `0x8B0F_090F`
- Spike: `0x8B00_090F`

The difference is in the value of `pmp2cfg` going being `0x0F` or `0x00`. The value that is being written there is `0x6F`, and is an illegal value because the sixth and fifth bit are supposed to be zero.

## Expected Behavior

I think Ibex does the correct thing because PMP configuration registers are WARL.

## Steps to reproduce the issue

Using Ibex DV with test `riscv_pmp_full_random_test` and seed 20233.

## More details
Ibex trace:
```
Time	Cycle	PC	Insn	Decoded instruction	Register and memory contents
           2249	        10	80000080	40001a37	lui	x20,0x40001	 x20=0x40001000
           2269	        11	80000084	106a0a13	addi	x20,x20,262	 x20:0x40001000 x20=0x40001106
           2289	        12	80000088	301a1073	csrrw	x0,misa,x20	 x20:0x40001106  x0=0x00000000
           2309	        13	8000008c	0002ce97	auipc	x29,0x2c	 x29=0x8002c08c
           2329	        14	80000090	c58e8e93	addi	x29,x29,-936	 x29:0x8002c08c x29=0x8002bce4
           2389	        17	80000094	00001a17	auipc	x20,0x1	 x20=0x80001094
           2729	        34	80000098	66ca0a13	addi	x20,x20,1644	 x20:0x80001094 x20=0x80001700
           2749	        35	8000009c	001a6a13	ori	x20,x20,1	 x20:0x80001700 x20=0x80001701
           2949	        45	800000a0	305a1073	csrrw	x0,mtvec,x20	 x20:0x80001701  x0=0x00000000
           2969	        46	800000a4	74725073	csrrwi	x0,0x747,4	  x0=0x00000000
           2989	        47	800000a8	00002217	auipc	x4,0x2	  x4=0x800020a8
           3009	        48	800000ac	0e020213	addi	x4,x4,224	  x4:0x800020a8  x4=0x80002188
           3029	        49	800000b0	4a01	c.li	x20,0	 x20=0x00000000
           3049	        50	800000b2	9252	c.add	x4,x20	  x4:0x80002188 x20:0x00000000  x4=0x80002188
           3109	        53	800000b4	00225213	srli	x4,x4,0x2	  x4:0x80002188  x4=0x20000862
           3129	        54	800000b8	3b021073	csrrw	x0,pmpaddr0,x4	  x4:0x20000862  x0=0x00000000
           3249	        60	800000bc	00002217	auipc	x4,0x2	  x4=0x800020bc
           3669	        81	800000c0	0cc20213	addi	x4,x4,204	  x4:0x800020bc  x4=0x80002188
           3789	        87	800000c4	6a05	c.lui	x20,0x1	 x20=0x00001000
           3929	        94	800000c6	3dfa0a13	addi	x20,x20,991	 x20:0x00001000 x20=0x000013df
           3949	        95	800000ca	9252	c.add	x4,x20	  x4:0x80002188 x20:0x000013df  x4=0x80003567
           3969	        96	800000cc	00225213	srli	x4,x4,0x2	  x4:0x80003567  x4=0x20000d59
           3989	        97	800000d0	3b121073	csrrw	x0,pmpaddr1,x4	  x4:0x20000d59  x0=0x00000000
           4009	        98	800000d4	00002217	auipc	x4,0x2	  x4=0x800020d4
           4029	        99	800000d8	0b420213	addi	x4,x4,180	  x4:0x800020d4  x4=0x80002188
           4089	       102	800000dc	6a11	c.lui	x20,0x4	 x20=0x00004000
           4109	       103	800000de	77ca0a13	addi	x20,x20,1916	 x20:0x00004000 x20=0x0000477c
           4129	       104	800000e2	9252	c.add	x4,x20	  x4:0x80002188 x20:0x0000477c  x4=0x80006904
           4289	       112	800000e4	00225213	srli	x4,x4,0x2	  x4:0x80006904  x4=0x20001a41
           4309	       113	800000e8	3b221073	csrrw	x0,pmpaddr2,x4	  x4:0x20001a41  x0=0x00000000
           4729	       134	800000ec	00002217	auipc	x4,0x2	  x4=0x800020ec
           4749	       135	800000f0	09c20213	addi	x4,x4,156	  x4:0x800020ec  x4=0x80002188
           5069	       151	800000f4	6a21	c.lui	x20,0x8	 x20=0x00008000
           5089	       152	800000f6	74ca0a13	addi	x20,x20,1868	 x20:0x00008000 x20=0x0000874c
           5109	       153	800000fa	9252	c.add	x4,x20	  x4:0x80002188 x20:0x0000874c  x4=0x8000a8d4
           5129	       154	800000fc	00225213	srli	x4,x4,0x2	  x4:0x8000a8d4  x4=0x20002a35
           5149	       155	80000100	3b321073	csrrw	x0,pmpaddr3,x4	  x4:0x20002a35  x0=0x00000000
           5169	       156	80000104	8b001237	lui	x4,0x8b001	  x4=0x8b001000
           5589	       177	80000108	d0520213	addi	x4,x4,-763	  x4:0x8b001000  x4=0x8b000d05
           5669	       181	8000010c	3a021073	csrrw	x0,pmpcfg0,x4	  x4:0x8b000d05  x0=0x00000000
           5689	       182	80000110	00002217	auipc	x4,0x2	  x4=0x80002110
           6069	       201	80000114	07820213	addi	x4,x4,120	  x4:0x80002110  x4=0x80002188
           6089	       202	80000118	6a25	c.lui	x20,0x9	 x20=0x00009000
           6109	       203	8000011a	ae5a0a13	addi	x20,x20,-1307	 x20:0x00009000 x20=0x00008ae5
           6129	       204	8000011e	9252	c.add	x4,x20	  x4:0x80002188 x20:0x00008ae5  x4=0x8000ac6d
           6149	       205	80000120	00225213	srli	x4,x4,0x2	  x4:0x8000ac6d  x4=0x20002b1b
           6169	       206	80000124	3b421073	csrrw	x0,pmpaddr4,x4	  x4:0x20002b1b  x0=0x00000000
           6309	       213	80000128	00002217	auipc	x4,0x2	  x4=0x80002128
           6729	       234	8000012c	06020213	addi	x4,x4,96	  x4:0x80002128  x4=0x80002188
           6749	       235	80000130	6a25	c.lui	x20,0x9	 x20=0x00009000
[... 365 lines cut ...]
core   0: 0x80000254 (0x74705073) csrwi   mseccfg, 0
core   0: 3 0x80000254 (0x74705073) c1863_mseccfg 0x00000000
core   0: 0x80000258 (0x189a5237) lui     tp, 0x189a5
core   0: 3 0x80000258 (0x189a5237) x4  0x189a5000
core   0: 0x8000025c (0x4cd20213) addi    tp, tp, 1229
core   0: 3 0x8000025c (0x4cd20213) x4  0x189a54cd
core   0: 0x80000260 (0x00002097) auipc   ra, 0x2
core   0: 3 0x80000260 (0x00002097) x1  0x80002260
core   0: 0x80000264 (0xf2808093) addi    ra, ra, -216
core   0: 3 0x80000264 (0xf2808093) x1  0x80002188
core   0: 0x80000268 (0x00009206) c.add   tp, ra
core   0: 3 0x80000268 (0x9206) x4  0x989a7655
core   0: 0x8000026a (0x3b021273) csrrw   tp, pmpaddr0, tp
core   0: 3 0x8000026a (0x3b021273) x4  0x20000862 c944_pmpaddr0 0x989a7655
core   0: 0x8000026e (0x3b021273) csrrw   tp, pmpaddr0, tp
core   0: 3 0x8000026e (0x3b021273) x4  0x989a7655 c944_pmpaddr0 0x20000862
core   0: 0x80000272 (0x676f3237) lui     tp, 0x676f3
core   0: 3 0x80000272 (0x676f3237) x4  0x676f3000
core   0: 0x80000276 (0x90f20213) addi    tp, tp, -1777
core   0: 3 0x80000276 (0x90f20213) x4  0x676f290f
core   0: 0x8000027a (0x3a021273) csrrw   tp, pmpcfg0, tp
core   0: 3 0x8000027a (0x3a021273) x4  0x8b000d05 c928_pmpcfg0 0x8b00090f
core   0: 0x8000027e (0x3a021273) csrrw   tp, pmpcfg0, tp
core   0: 3 0x8000027e (0x3a021273) x4  0x8b00090f c928_pmpcfg0 0x8b000d05
```

ctopal, 2022-08-16:
This is fixed in tag ibex-cosim-v0.4

closed 2022-08-16 by ctopal

marnovandermaas, 2022-08-16:
Thanks @ctopal!