lowRISC/ibex #1787: [dv] More flexible riscv_core_setting.sv generation

GregAC, 2022-08-18:
Currently the DV flow patches the `riscv_core_setting.sv` file if certain configs are used to remove CSRs those configs do not have and adjust PMP settings.

We need something more flexible that can look directly at the Ibex config and generate the file based upon settings in there.

In particular we need this to enable https://github.com/lowRISC/ibex/pull/1776. With better CSR coverage we're seeing failures in different configs due to different CSR setups.

I would suggest we an existing python templating library (e.g. Jinja2). We'll need a small python script that can read in the ibex config and pass it through to jinja2 to produce the final `riscv_core_setting.sv`.

@marnovandermaas @ctopal @hcallahan-lowrisc any thoughts?

(Estimate based on us following my idea above)

> estimate 4

GregAC, 2022-08-18:
We could maybe extend the current patch setup to do more things using more patch files but it quickly becomes brittle. So I think this is something that's worth doing for V2 (may end up more time juggling patch files later on to hit V2 as we discover other things the templating setup becomes useful for).

Another option would just be to cease running CI against anything but the opentitan config, but I do not like that idea.

marnovandermaas, 2022-08-18:
Yes, I added the patch mechanism when I enabled the PMP tests, but it was definitely meant as a short-term solution as long as that was the only check that needed to happen. I've never used this particular templating library, link for reference:
https://jinja.palletsprojects.com/en/3.1.x/
But it looks like a good option so that we only have to specify the differences between the different configurations (which was the original goal of the patch system as well).

hcallahan-lowrisc, 2022-08-19:
@marnovandermaas and I talked about templating as an option when discussing PMP stuff a few weeks back. I think it makes the most sense here. Maybe we should use Mako as that is already used within the Ibex project (it's already in the python_requirements.txt), though I wouldn't be against using Jinja2 either.

A related comment I have is that it does feel a bit confusing with options to riscv-dv being passed through both the core_settings file, and on the cmd line. Plus the other yaml files. Eg. you set `supported_isa[$]` in core_setting.sv and also pass `--isa`  to riscvdv/run.py. Though that's a discussion for another time I think.

marnovandermaas, 2022-08-19:
@hcallahan-lowrisc good idea about Mako, I'll paste the reference here for convenience: https://www.makotemplates.org/

GregAC, 2022-08-19:
> Maybe we should use Mako as that is already used within the Ibex project (it's already in the python_requirements.txt)

Ah yes we should. For some reason I thought we were using jinja2 within OpenTitan but we're not it's Mako everywhere.

> A related comment I have is that it does feel a bit confusing with options to riscv-dv being passed through both the core_settings file, and on the cmd line. Plus the other yaml files. Eg. you set [...]

Definitely, on both points. A complete refactor to make configuration more sane is needed but not something we want to look at now.

commit 3d763006 (2022-08-30): [dv] Add makefile step for generating core config file from template

The prior system of using a patch to alter the riscv_dv_setting.sv file
has been removed and replaced with a mako templating based approach.

Fixes #1787
closed 2022-08-30 by GregAC