lowRISC/ibex #1789: [dv,riscvdv] PMP full random test stopping after first fault

marnovandermaas, 2022-08-19:
In the PMP full random test, I would like to exercise the random PMP configuration with multiple stores and load. However, the current RISCV-DV  exception handler simply exits the test if the PMP entry is locked at all, which is quite likely with MML mode. Instead I would like the test to continue after seeing a fault to the next load/store.

> estimate 16

marnovandermaas, 2022-09-02:
A related issue that also has to do with the PMP trap handler is that when a PMP region is set to be non-executable. The trap handler just tries to continue with the next instruction, which makes for very slow execution (the trap handler is called for every instruction). This inevitably leads to time-outs. I think in case there is an instruction access fault due to the PMP, we should just stop the test.

marnovandermaas, 2022-10-05:
The main issue with the proposed solution is that it is not clear whether the instruction at `mepc` is compressed or not. Both default increments of 2 and 4 bytes will have issues because Ibex supports misaligned instruction addresses.

marnovandermaas, 2022-10-10:
I made the MML code entry execute and read by default to be able to continue after a store/load. In the load trap handler I check whether `mepc` was before main, as a way to check whether there was a load access fault inside the trap handler.

I have have a branch for RISCV-DV that has these changes, but I want to rope in https://github.com/lowRISC/ibex/issues/1742 before creating a PR

> spent 8

marnovandermaas, 2022-10-10:
My changes need some more testing so I am going to up the estimate from 16 to 24 so that I have an extra 8 to work on this.

marnovandermaas, 2022-10-13:
There is a PR for this now: https://github.com/google/riscv-dv/pull/903

GregAC, 2022-10-25:
@hcallahan-lowrisc to merge @marnovandermaas's Ibex changes

GregAC, 2022-11-01:
With https://github.com/lowRISC/ibex/pull/1876 this test is now behaving reasonably

closed 2022-11-01 by GregAC