lowRISC/ibex #1804: No such file or directory 'xrun' when trying to make the testbench.

AshvinVaidyanathan, 2022-09-07:
I am trying to run the testbench using the make file in /dv/uvm/core_ibex using the commands in the documentation, but I get the error:

Errno 2] No such file or directory: 'xrun'
make: *** [Makefile:319: out/metadata/tb.compile.stamp] Error 1

I am unable to get anything about what this xrun file is or where it is located. I understand that it is being caused due to incorrect paths set but I can't seem to figure out where the error is.

GregAC, 2022-09-07:
What simulator do you want to use? `xrun` is part of xcelium (from Cadence) which is the default simulator. If you don't have access to xcelium then you cannot run the testbench with the default setup. Other simulators can be chosen with the `SIMULATOR=` option passed to the make command.  

You can see the set of supported simulators in this file: https://github.com/lowRISC/ibex/blob/master/dv/uvm/core_ibex/yaml/rtl_simulation.yaml though beware only Xcelium and VCS are regularly tested by us.

AshvinVaidyanathan, 2022-09-07:
I am using the SIMUALTOR=verilator, which gives the above error. Though now that I look at the yaml file I don't see a verilator option. But the System and Tool Requirements section of the docs says Verilator is supported. Do I need to follow some extra steps for this?

GregAC, 2022-09-07:
Unfortunately we do not support verilator with the UVM testbench. Verilator doesn't support the parts of the system verilog language required to run it (though such support is being developed).

Verilator can be used to run a CPU simulation (like simple system), and can be used with the co-simulation setup, but we it doesn't give you the randomised stimulus and randomly generated programs you get with the main testbench.

AshvinVaidyanathan, 2022-09-07:
I see. Thanks a lot for the clarification.

closed 2022-09-07 by AshvinVaidyanathan