lowRISC/ibex #1809: Question about misaligned memory requests that cause bus errors

ivanmgribeiro, 2022-09-19:
Hi,
I would like to ask what the intended behaviour of a system with an Ibex core is when a load/store instruction with a misaligned address leads to the first memory request causing a bus error, but the second request _not_ causing a bus error.

For example, assume the memory map of the system contains an unmapped region between `0x0000` and `0x7FFF`, followed by some RAM starting at address `0x8000`, and the bus generates a bus error signal when the address of a request is not mapped.

If software attempts a word-width store to address `0x7FFF`, the Ibex core will generate a memory request to address `0x7FFC` to write only the upper byte. Once this first request is granted, the LSU transitions into the `WAIT_RVALID_MIS` state, in which it unconditionally issues the next request, regardless of whether the first caused an error or not. This second request will be to address `0x8000`, which in this example is mapped to RAM. As far as I can tell, from a bus protocol point of view the second half of the store instruction is treated as an entirely separate bus request, so it makes it to the RAM that is mapped and writes to the underlying memory.

In this case, a store that leads to an exception has still written to the underlying memory, which should not happen.

I would expect in this case that the second bus request should only be made when the first has been confirmed to not cause an error. Is this the correct behaviour to expect or am I misunderstanding something?

I ran into this while using a modified version of the Ibex core (although the section that handles bus errors has not been changed). I didn't see an easy way to replicate this using the unchanged code since the example systems in the repository don't lead to errors when an unmapped address is requested. However, as far as I can see from looking through the LSU file it looks like the current behaviour matches the behaviour described above.

Thanks,
Ivan

marnovandermaas, 2022-09-28:
From the use spec (pg 19) misaglined stores and loads are not guaranteed to be atomic. In the privileged spec on PMP (pg 60): "In particular, a portion of a misaligned store that passes the PMP check may become visible, even if another portion fails the PMP check." I think the same is probably true for unmapped regions, in which case the behavior you are describing follows the spec.

@GregAC do you also agree that parts of misaligned stores are allowed to succeed even if other parts cause an access fault? The other question is whether we want to stop the side effect in this particular case.

GregAC, 2022-10-03:
There's a couple of reasons to leave things as the way they are (second half of an misaligned access allowed to send a request before we know the error status of the first):

1. Performance - If you have a few cycles latency on the error response but can send out multiple requests this allows you to get both requests out quicker
2. Reducing coupling/feedthroughs - If we did wait for the first request to finish before deciding whether to start the second we'd need to have a direct path from the data response in to the data request out (or reduce performance further but not starting the second request until the cycle after the first is done). As it happens we already have such a path (to enable back to back memory accesses) so this isn't a big problem but still minimising such paths is a good idea

Then given the above 2 a final reason/question:

What does waiting on the success of the first before allowing the second give you? What it doesn't give you is an architectural guarantee that either the entirety of a misaligned store happens or none of it happens as you still have the case where the second half fails when the first succeeds.

ivanmgribeiro, 2022-10-03:
Thanks to you both for the insightful responses. I wasn't aware that the RISC-V specification specifically allowed for only part of a misaligned load or store to succeed. The points you raised make sense.

I also only saw the discussion in https://github.com/lowRISC/opentitan/issues/1868 and https://github.com/lowRISC/ibex/issues/764 recently.

closed 2022-10-03 by ivanmgribeiro