lowRISC/ibex #1810: Performance Counters doesn't work.

weiyan-zhang, 2022-09-20:
## My Environment

**EDA tool and version:** Verilator 4.226 and spike-cosim

**Operating system:** Ubuntu

**Version of the Ibex source code:** 2f8dfa9dfe2d938228118ed8ab6a543804d0c9ae
  

If I run the co-simulation, the performance counters looks like this. All counters are 0 except Cycles and Instructions Retired.

```
build/lowrisc_ibex_ibex_simple_system_cosim_0/sim-verilator/Vibex_simple_system --meminit=ram,examples/sw/benchmarks/coremark/coremark.elf
```
Performance Counters
====================
Cycles:                     4009545
NONE:                       0
Instructions Retired:       2725247
LSU Busy:                   0
Fetch Wait:                 0
Loads:                      0
Stores:                     0
Jumps:                      0
Conditional Branches:       0
Taken Conditional Branches: 0
Compressed Instructions:    0
Multiply Wait:              0
Divide Wait:                0

GregAC, 2022-09-21:
The first thing to check is how did you build the simulation? You may have configured Ibex to have no performance counters, hence the results.

weiyan-zhang, 2022-09-22:
I tried * *simple system* * following https://github.com/lowRISC/ibex/blob/master/examples/sw/benchmarks/README.md
and * *co-simulation system* * following https://ibex-core.readthedocs.io/en/latest/03_reference/cosim.html

jiangnanguqi, 2022-10-27:
Hi,

I just solved this.
Please check
https://github.com/lowRISC/ibex/issues/1866#issuecomment-1293541290

WeiyanZh, 2022-10-29:
It works, thank you!

vogelpi, 2023-01-16:
Closing the issue as the question has been answered and the problem could be solved.

closed 2023-01-16 by vogelpi