lowRISC/ibex #1823: [dv] Implement double fault loop detector

GregAC, 2022-09-29:
Add a double fault loop detector to the testbench. It should determine when Ibex has entered a loop of double faults and terminate the simulation after some configurable number of iterations (with a default e.g. 100).

By default is should exit with a fail but a configuration option should be supplied to enable exit with a pass (for tests where a double fault loop is the intended behaviour).

> estimate 4

GregAC, 2022-10-06:
Implementation here: https://github.com/lowRISC/ibex/pull/1808/commits but not yet merged

marnovandermaas, 2022-10-12:
I am encountering these double faults (or faults that happen when handling another fault) during some PMP tests. To get an infinte loop of instruction access fault exceptions you can do the following.
Clone this repo and branch: https://github.com/marnovandermaas/ibex/tree/catch-dv-double-faults
Specifically use this commit: https://github.com/marnovandermaas/ibex/tree/10f561a4ddc8a381583fe0d63b9fae23f9b3b003

And then run the following command inside `dv/uvm/core_ibex`:
```
make -j1 --keep-going IBEX_CONFIG=opentitan ITERATIONS=1 TEST=riscv_pmp_full_random_test WAVES=0 COV=0 SEED=17397
```

hcallahan-lowrisc, 2022-10-12:
> remaining 2022-10-12 1

hcallahan-lowrisc, 2022-10-12:
Stil TODO:
- Add option to change the test result from a pass to a fail when the detector activates.

PR #1848 (2022-10-17): Add a double_fault detector to core_ibex uvm environment
Add a new scoreboard component to the core_ibex uvm environment, which contains a double_fault detector task. This uses the top-level output 'double_fault_seen_o' to count the number of total and consecutive double_faults seen within a test. A helper task allows the base_test to wait upon each of these counters reaching the configured thresholds, and then to end the test.

The default thresholds are 100 for consecutive faults, and 1000 for total faults.
When either threshold is reached, the test will fail by default.

A plusarg '+enable_ [... cut]
closed 2022-10-17 by hcallahan-lowrisc