lowRISC/ibex #1825: [dv] Implement testing and coverage for various mstatus states

GregAC, 2022-09-30:
I recently discovered Ibex enabled/disabled interrupts in U mode based on the mstatus.mie bit which is a violation of the spec (mie applies only to M-mode in U mode it is ignored, only the per interrupt MIE CSR bits should be considered. This is fixed in https://github.com/lowRISC/ibex/pull/1824

I found this when adding interrupts to the debug single step test, which does execute code in U mode with mstatus.mie set to 0, but clearly other testing (in particular the interrupt tests) don't do this otherwise we would have found this bug earlier.

This indicates a gap in our testing. We should add some coverage over different mstatus states that checks we see various states across U-mode/M-mode and with interrupts/debugs/exceptions occurring.

Part of this task is producing a more concrete definition of the above paragraph.

Existing tests will require expanding to meet the new coverage

> estimate 4

GregAC, 2022-11-01:
Consider waiving this work for V2, postponing to V3 to discuss in V2 meeting.

johngt, 2022-11-16:
I took a quick look at both review meeting notes and couldn't see an obvious waiver discussion on this so assume that this is still live?

GregAC, 2022-11-16:
@johngt This has been waived for V2, I've added a comment to the notes tagging you to highlight where

GregAC, 2023-04-19:
The suggested crosses have been added and are almost entirely covered.

The gap is we're not seeing internal NMIs (caused on ECC errors from incoming load data) ocurring in U mode when interrupts are globally enabled (mstatus.mie == 1). This is due to the way to test works and should be simple enough to fix which I'll do for M2.5.

GregAC, 2023-04-19:
Coverage hole should be fixed by https://github.com/lowRISC/ibex/pull/2017

GregAC, 2023-06-21:
Coverage holes are addressed by latest DV, though we need to up the interations of the riscv_mem_intg_error_test to reliably hit it all every time: https://github.com/lowRISC/ibex/pull/2051

closed 2023-06-22 by GregAC