lowRISC/ibex #1826: python flow error in dv/uvm/core_ibex/scripts/riscvdv_interface.py

SerLippo, 2022-10-03:
After I use command `make TEST=riscv_machine_mode_rand_test ITERATIONS=1 SIMULATOR=vcs`, terminal just report this:
```
Build metadata already exists, not recreating from scratch.
Building RTL testbench
Parameters in riscvdv command not substituted!
Parameters : ['<rtl_log>']
Command :  vcs
  -full64
  -f /home/ser_lip/Projects/ibex/ibex/dv/uvm/core_ibex/ibex_dv.f
  -l /home/ser_lip/Projects/ibex/ibex/dv/uvm/core_ibex/out/build/tb/<rtl_log>
  -sverilog
  -ntb_opts uvm-1.2
  +define+UVM
  +define+UVM_REGEX_NO_DPI
  -timescale=1ns/10ps
  -licqueue
  -LDFLAGS '-Wl,--no-as-needed'
  -CFLAGS '--std=c99 -fno-extended-identifiers'
  -Mdir=/home/ser_lip/Projects/ibex/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv.csrc
  -o /home/ser_lip/Projects/ibex/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv
  -debug_access+pp
  -xlrm uniq_prior_final
  -CFLAGS '--std=c99 -fno-extended-identifiers'
  -lca -kdb
  -pvalue+core_ibex_tb_top.RV32E=0 +define+IBEX_CFG_RV32M=ibex_pkg::RV32MSingleCycle +define+IBEX_CFG_RV32B=ibex_pkg::RV32BOTEarlGrey +define+IBEX_CFG_RegFile=ibex_pkg::RegFileFF -pvalue+core_ibex_tb_top.BranchTargetALU=1 -pvalue+core_ibex_tb_top.WritebackStage=1 -pvalue+core_ibex_tb_top.ICache=1 -pvalue+core_ibex_tb_top.ICacheECC=1 -pvalue+core_ibex_tb_top.ICacheScramble=1 -pvalue+core_ibex_tb_top.BranchPredictor=0 -pvalue+core_ibex_tb_top.DbgTriggerEn=1 -pvalue+core_ibex_tb_top.SecureIbex=1 -pvalue+core_ibex_tb_top.PMPEnable=1 -pvalue+core_ibex_tb_top.PMPGranularity=0 -pvalue+core_ibex_tb_top.PMPNumRegions=16 -pvalue+core_ibex_tb_top.MHPMCounterNum=10 -pvalue+core_ibex_tb_top.MHPMCounterWidth=32   -f /home/ser_lip/Projects/ibex/ibex/dv/uvm/core_ibex/ibex_dv_cosim_dpi.f -LDFLAGS '-Wl,-rpath,/opt/riscv/ibex_toolchain/lib' -CFLAGS '-I/opt/riscv/ibex_toolchain/include -I/opt/riscv/ibex_toolchain/include/softfloat -I/opt/riscv/ibex_toolchain/include' -CFLAGS '-I/home/ser_lip/Projects/ibex/ibex/dv/cosim' -L/opt/riscv/ibex_toolchain/lib -lriscv -lsoftfloat -ldl -ldisasm -lfdt -lstdc++

Traceback (most recent call last):
  File "/home/ser_lip/Projects/ibex/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 108, in <module>
    sys.exit(_main())
  File "/home/ser_lip/Projects/ibex/ibex/dv/uvm/core_ibex/scripts/compile_tb.py", line 85, in _main
    md.tb_build_cmds = riscvdv_interface.get_tool_cmds(
  File "/home/ser_lip/.local/lib/python3.10/site-packages/typeguard/__init__.py", line 1033, in wrapper
    retval = func(*args, **kwargs)
  File "/home/ser_lip/Projects/ibex/ibex/dv/uvm/core_ibex/scripts/riscvdv_interface.py", line 165, in get_tool_cmds
    raise RuntimeError
RuntimeError
```
I find that the `<rtl_log>` is not substituted by the real file name. 

## My Environment

Just setup the latest environment.
latest ibex master branch with no edit. `3943a4eca3917d0a00406abb36f4951304c0b394`
Spike. ibex-cosim-v0.3
Toolchain. lowrisc-toolchain-rv32imcb-20220524-1 release.

**EDA tool and version:**

Synopsys VCS 2018.09

**Operating system:**

Ubuntu Linux Desktop 22.04

**Version of the Ibex source code:**

latest ibex master branch with no edit. 
```
3943a4eca3917d0a00406abb36f4951304c0b394
Author: Marno van der Maas <mvdmaas+git@lowrisc.org>
Date:   Tue Sep 27 11:36:50 2022 +0100

    [pmp] Remove off mode from pmp_*_mode_cross coverpoints
```

SerLippo, 2022-10-05:
By the way, my python is 3.10.

ctopal, 2022-10-05:
Hey @SerLippo, thanks for reporting this! We are currently working on a fix to make VCS work again with our DV environment. We were not using it for a while and it got broken, sorry about that.

I've opened up a fix on riscv-dv repository to make it compile with VCS in here https://github.com/google/riscv-dv/pull/901. And also this branch here https://github.com/lowRISC/ibex/tree/vcs_fix for Ibex on top of that fix should do the trick. Note that this branch will turn into a PR and eventually will be merged into master with the vendored riscv-dv but it might take couple of days.

Please let us know if you encounter any more problems!
-C

closed 2022-10-17 by ctopal