lowRISC/ibex #1827: `core_busy_o` not protected by lockstep

andreaskurth, 2022-10-03:
## Observed Behavior
The `core_busy_o` output signal of the non-lockstep core is used to gate the core clock, which is used by the regular Ibex core and its lockstep counterpart.
https://github.com/lowRISC/ibex/blob/3943a4eca3917d0a00406abb36f4951304c0b394/rtl/ibex_top.sv#L211-L219
https://github.com/lowRISC/ibex/blob/3943a4eca3917d0a00406abb36f4951304c0b394/rtl/ibex_top.sv#L286-L287
https://github.com/lowRISC/ibex/blob/3943a4eca3917d0a00406abb36f4951304c0b394/rtl/ibex_top.sv#L927-L928

This means that the lockstep core gets gated in the same clock cycle as the regular core, even though the lockstep core is a few cycles delayed. **A low-glitch in the `core_busy_o` output of the non-lockstep core thus does not lead to a comparison mismatch and a major alert because the lockstep core is already clock-gated when it would produce the value of `core_busy_o` that is used in the comparison.**
https://github.com/lowRISC/ibex/blob/3943a4eca3917d0a00406abb36f4951304c0b394/rtl/ibex_lockstep.sv#L453-L454

## Expected Behavior

A low-glitch/mismatch in the `core_busy_o` output signal of the non-lockstep core should lead to a major alert by the lockstep core.

## Steps to reproduce the issue
Run the `chip_sw_ibex_lockstep_glitch` test in OpenTitan (currently in https://github.com/lowRISC/opentitan/pull/15221) and override `port_idx` to the index of `core_busy_o` in `output_ports`.

vogelpi, 2022-10-04:
Thanks @andreaskurth for creating this issue. I think this is a good catch!

I've now had a look at the RTL. If the clock is turned off, also the delay registers have no clock anymore. This means the outputs of the last couple of cycles of the core will be compared with the outputs of the lockstep/shadow core after resuming. Functionality-wise this seem okay to me.

From a security perspective, this could however be problematic. As we just discussed in person, by glitching `core_busy_o` of the main core to low, Ibex could be made to sleep instead of handling e.g. an alert or similar. Eventually, the watchdog timer will intervene and reset the whole chip but Andreas and I think it would be good to just protect the `core_busy_o` signal using a multi-bit encoding (similar to `fetch_enable_i`). Does this sound reasonable @GregAC ?

An alternative would be to only disable the clock once also the lockstep/shadow core outputs `core_busy_o` low. But this would probably require invasive changes to the core controller currently expects the clock to be switched off one clock cycle after setting busy low. I think the pipeline can not be made to not fetch instructions when the clock is on. Maybe, one could use the `fetch_enable_i` mechanism for the purpose but I am not sure we want to do that.

GregAC, 2022-10-05:
> Eventually, the watchdog timer will intervene and reset the whole chip but Andreas and I think it would be good to just protect the core_busy_o signal using a multi-bit encoding (similar to fetch_enable_i). Does this sound reasonable @GregAC ?

I think so.

As you point out actually delaying the clock enable for the lockstep has it's own issues so hardening the signal instead makes sense. I think there's also limited scope for such a glitch attack on `core_busy_o` to achieve much as whilst you may be able to skip an alert you'll need to let the CPU start up again at some point at which point there will be an output mismatch (as the lockstep core won't produce a matching `core_busy_o` drop). So this helps justify a different approach here for the `core_busy_o` output.

GregAC, 2022-10-05:
@msfschaffner @tjaychen @cdgori any thoughts?

cdgori, 2022-10-05:
I think multi-bit hardening of `core_busy_o` is reasonable here, because as you say, when the CPU ultimately starts up again there will be an output mismatch.

Can anyone brainstorm a way for an adversary to suppress that mismatch?  I'm drawing a blank.  That would be the only reason to think about the more invasive change to combine the two `core_busy_o` signals to form the clock enable.

vogelpi, 2022-10-11:
Thanks for the feedback @cdgori , I couldn't come up with a scenario for someone to suppress the mismatch.

I will have a look at doing the RTL modification for this.

msfschaffner, 2022-10-12:
The MUBI solution SGTM too!

closed 2022-10-25 by vogelpi