lowRISC/ibex #1843: [dv] Prettify invalid option to column

marnovandermaas, 2022-10-07:
## Observed Behavior

When running `make prettify` I get the following error:
```
column: invalid option -- 'o'
usage: column [-txne] [-c columns] [-s sep] [file ...]
```

## Expected Behavior

I expect that the content of `scripts/prettify.sh` contains the options that are compatible on my system or otherwise tell me which one to use.

## My Environment

**Operating system:**

Ubuntu Linux 20.04

The manual for my column command states "BSD General Commands Manual" from July 29, 2004

**Version of the Ibex source code:**

7ab2571bea0daca3f14155d82f0124fb0b118799

hcallahan-lowrisc, 2022-10-14:
For me (Ubuntu Linux 22.04), column reports its version as `util-linux 2.37.2` (2021-06-02) in both the CLI and manpage. Must be a newer version that introduced `-o` I guess.

SamuelRiedel, 2026-10-07:
Fixes on new OS.

closed 2026-10-07 by SamuelRiedel