lowRISC/ibex #1860: Problem on Building Simulation with RV32E

sadjadtu, 2022-10-18:
I am trying to build simulation, but I got this error:

ERROR: ../src/lowrisc_ibex_ibex_simple_system_core_0/ibex_simple_system.cc: In member function ‘std::__cxx11::string SimpleSystem::GetIsaString() const’:
../src/lowrisc_ibex_ibex_simple_system_core_0/ibex_simple_system.cc:40:46: error: ‘class Vibex_simple_system_ibex_simple_system’ has no member named ‘RV32E’; did you mean ‘RV32M’?
   std::string base = top.ibex_simple_system->RV32E ? "rv32e" : "rv32i";

I am using Ubuntu Linux 18.04

sadjadtu, 2022-10-19:
I fixed it. The problem was about not properly cloning them.

jiangnanguqi, 2022-10-19:
> I fixed it. The problem was about not properly cloning them.

Hi,
How did you fix it? I am suffering from the same problem when I build the simulation.
I have tried to change "RV32E" to "RV32M" in the ibex_simple_system.cc file. 
Then it can be built without errors and produce the expected "hello_test" result.
But I don't think this is a proper solution as it ignores the E extension.

sadjadtu, 2022-10-19:
> 
Follow the instruction explained in this website:

https://blog.csdn.net/qq_33766313/article/details/112780529

In my case, I needed to start over again the setting up.

enochliang, 2022-10-24:
> I fixed it. The problem was about not properly cloning them.

What files are not properly cloned?
I am suffer from it too. And I followed the instructions in the website you gave. But the error still exist.

jiangnanguqi, 2022-10-25:
> 

Hi,

My temporary solution is
In this file ../ibex/example/simple_system/ibex_simple_system.cc:40
Modify  "std::string base = top.ibex_simple_system->RV32E ? "rv32e" : "rv32i";"
To "std::string base = "rv32i";"
which means just to skip the setting of RV32E
Then the building will be successful without errors.

But I am still looking for better solutions to reserve the setting of RV32E

jiangnanguqi, 2022-10-25:
> > I fixed it. The problem was about not properly cloning them.
> 
> What files are not properly cloned? I am suffer from it too. And I followed the instructions in the website you gave. But the error still exist.

Hi,

My temporary solution is
In this file ../ibex/example/simple_system/ibex_simple_system.cc:40
Modify "std::string base = top.ibex_simple_system->RV32E ? "rv32e" : "rv32i";"
To "std::string base = "rv32i";"
which means just to skip the setting of RV32E
Then the building will be successful without errors.

But I am still looking for better solutions to reserve the setting of RV32E

enochliang, 2022-10-27:
I found that the problem is in the verilator installation flow.
I typed "apt-get install verilater" to install verilator before, now I follow this instructions to install it.
Maybe using apt-get to do a simple installation cannot build verilator very well.

```
# Prerequisites:
#sudo apt-get install git perl python3 make autoconf g++ flex bison ccache
#sudo apt-get install libgoogle-perftools-dev numactl perl-doc
#sudo apt-get install libfl2  # Ubuntu only (ignore if gives error)
#sudo apt-get install libfl-dev  # Ubuntu only (ignore if gives error)
#sudo apt-get install zlibc zlib1g zlib1g-dev  # Ubuntu only (ignore if gives error)

git clone https://github.com/verilator/verilator   # Only first time

# Every time you need to build:
unsetenv VERILATOR_ROOT  # For csh; ignore error if on bash
unset VERILATOR_ROOT  # For bash
cd verilator
git pull         # Make sure git repository is up-to-date
git tag          # See what versions exist
#git checkout master      # Use development branch (e.g. recent bug fixes)
#git checkout stable      # Use most recent stable release
#git checkout v{version}  # Switch to specified release version

autoconf         # Create ./configure script
./configure      # Configure and create Makefile
make -j `nproc`  # Build Verilator itself (if error, try just 'make')
sudo make install
```

And the rest of environment building instructions you can follow the instructions in this website.[https://blog.csdn.net/qq_33766313/article/details/112780529](url) 

Now I can do a simulation well.

KatCe, 2023-07-04:
We encountered the same problem when using Verilator provided by apt-get in Ubuntu 20.04.5: Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad. Fixed the problem by compiling and using Verilator version 4.210