lowRISC/ibex #1863: insn_trigger_exception coverpoint does not work

hcallahan-lowrisc, 2022-10-19:
The test riscv_debug_triggers_test added in #1801 generates stimulus to hit [this coverpoint](https://github.com/lowRISC/ibex/blob/ce536ae47677e952de02b7421e1b78efb5761ab2/dv/uvm/core_ibex/fcov/core_ibex_fcov_if.sv#L502), but it does not trigger.

REPRO
```
// Checkout https://github.com/lowRISC/ibex/pull/1801
make TEST=riscv_debug_triggers_test ITERATIONS=1 SEED=14664 COV=1 -j12 -ik
// The instruction/event that should trigger this coverpoint can be seen retiring at t = 1010124 ns
```

hcallahan-lowrisc, 2022-10-24:
```verilog
  // > ibex_cs_registers.sv
  // L1560
   assign trigger_match[i] = tmatch_control_q[i] & (pc_if_i[31:0] == tmatch_value_q[i]);

   // > core_ibex_fcov_if.sv
   // L297
   // '_q' is just flopped one cycle later
   assign instr_id_matches_trigger_d = id_stage_i.controller_i.trigger_match_i &&
                                       id_stage_i.controller_i.fcov_debug_entry_if;
   // L502
   DV_FCOV_EXPR_SEEN(insn_trigger_exception, instr_id_matches_trigger_q &&
                                             id_stage_i.controller_i.fcov_pipe_flush)

   // > ibex_controller.sv
   // L908
   `DV_FCOV_SIGNAL(logic, debug_entry_if, (ctrl_fsm_cs != DBG_TAKEN_IF) &
                                          (ctrl_fsm_ns == DBG_TAKEN_IF))
   // L912
   `DV_FCOV_SIGNAL(logic, pipe_flush, (ctrl_fsm_cs != FLUSH) &
                                      (ctrl_fsm_ns == FLUSH))
```

Due to the hardware trigger `trigger_match_i` signal matching from the `pc_if`,  we see this match before the instruction has been decoded, and jump to DGB_TAKEN_IF, setting debug_mode = '1. Hence, decoding the instruction that creates the match in the if_stage (eg. illegal instruction) does not cause the controller to go to FLUSH, and the fcov_pipe_flush is never true in this case.

hcallahan-lowrisc, 2022-10-24:
@GregAC Do you have any cycles to take a look at this? Thanks

GregAC, 2022-10-25:
Turns out this coverpoint makes no sense. When we have a trigger point set on an instruction we stop execution just before that instruction executes, so that instruction will never take an exception. Instead let's have a single step over instruction that takes an exception coverpoint.

This is dealt with in #1872

closed 2022-10-26 by GregAC