lowRISC/ibex #1866: Performance Counters doesn't work

sadjadtu, 2022-10-21:
When I run the Simulator for the existing simple example, I get these results:

Simulation statistics
=====================
Executed cycles:  13142
Wallclock time:   0.024 s
Simulation speed: 547583 cycles/s (547.583 kHz)

Performance Counters
====================
Cycles:                     483
NONE:                       0
Instructions Retired:       266
LSU Busy:                   0
Fetch Wait:                 0
Loads:                      0
Stores:                     0
Jumps:                      0
Conditional Branches:       0
Taken Conditional Branches: 0
Compressed Instructions:    0
Multiply Wait:              0
Divide Wait:                0

I have read the similar questions, but I cannot find out what exactly I should do to resolve these wrong zeros.

jiangnanguqi, 2022-10-27:
Hi,

I just found the solution from here:
https://ibex-core.readthedocs.io/en/latest/03_reference/performance_counters.html

Try to add "--MHPMCounterNum=10" when building the simple system.
For example:
~/ibex$ fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --MHPMCounterNum=10

As mentioned in the Reference Guide, the Number of MHPMcounter is set to 0 as default, which means none of these counters are active excepet mcycle and mistret.

sadjadtu, 2022-10-29:
It works. Thank you

closed 2023-01-16 by vogelpi