lowRISC/ibex #1869: PMP Top Level Coverage Holes

ctopal, 2022-10-24:
- MPRV Effect cross is not getting hit for the cases: 
  - [ ] Privilege U->M Error 0 -> 1
  - [ ] Privilege U->M Error 1 -> 0
  - [ ] Privilege M->U Error 1 -> 0
 
Note that `->` stands for MPRV changing LSU privilege mode.

- Sticky Bit Features for:
  - [ ] RLB
  - [ ] MMWP
  - [ ] MML

- Data Channel:
  - [ ] No PMP region match && `mseccfg.mmwp` set.

- Instruction Channel 
  - [ ] No PMP region match && `mseccfg` register values other than 0.
  - [ ] Matching PMP region with NA4 mode (both allow and deny missing)

ctopal, 2022-10-24:
- For Instruction channel:
  Plan for now is to work on a directed instruction stream to inject to a new PMP test (possibly named pmp_directed) to have a high priority pmpcfg register set as NA4 address mode with pmpaddr pointing to a executable code region. That could also write some random `mseccfg` values while running to cover some more holes.

ctopal, 2022-10-28:
>remaining 2022-10-28 4

GregAC, 2023-03-23:
Covered by estimate in https://github.com/lowRISC/ibex/issues/1944 so removing estimates from here.

GregAC, 2023-06-22:
Closing this to leave #1944 for remaining tracking.

closed 2023-06-22 by GregAC