lowRISC/ibex #1920: Ask a question related to Ibex

msfschaffner, 2022-11-09:
https://ibex-core.readthedocs.io/en/latest/03_reference/exception_interrupts.html#internal-interrupts claims that 
```
Load integrity error internal interrupt. Only generated when SecureIbex == 0. mtval gives the faulting address. The interrupt will be taken at most one instruction after the faulting load. In particular a load or store immediately after a faulting load may execute before the interrupt is taken.
```
Isn't it the other way round - i.e., isn't the error only generated when `SecureIbex == 1`?

GregAC, 2022-11-10:
> Isn't it the other way round - i.e., isn't the error only generated when SecureIbex == 1?

Correct, I'll open an PR to fix the typo

vogelpi, 2022-11-10:
Thanks @GregAC !

msfschaffner, 2022-11-11:
Another question: is this table [here](https://ibex-core.readthedocs.io/en/latest/03_reference/exception_interrupts.html#exceptions) missing the `InstrMisalinged` exception?

GregAC, 2022-11-29:
> Another question: is this table [here](https://ibex-core.readthedocs.io/en/latest/03_reference/exception_interrupts.html#exceptions) missing the `InstrMisalinged` exception?
> 
> We seem to be defining this in ibex.h on the OpenTitan side:
> 
> ```c
> /**
>  * An Ibex exception type.
>  *
>  * This enum is used to decode RISC-V exception causes generated by Ibex.
>  */
> typedef enum ibex_exc {
>   kIbexExcInstrMisaligned = 0,
>   kIbexExcInstrAccessFault = 1,
>   kIbexExcIllegalInstrFault = 2,
>   kIbexExcBreakpoint = 3,
>   kIbexExcLoadAccessFault = 5,
>   kIbexExcStoreAccessFault = 7,
>   kIbexExcUserECall = 8,
>   kIbexExcMachineECall = 11,
>   kIbexExcMax = 31
> } ibex_exc_t;
> ```

No it isn't, you cannot generate instruction misaligned exceptions when you have the 'C' extension. I've fixed the ``IbexSecure`` issue and added a clarification for the instruction misaligned exception here: https://github.com/lowRISC/ibex/pull/1957

msfschaffner, 2022-11-29:
Great, thanks for the clarification!

closed 2022-12-08 by GregAC