lowRISC/ibex #1933: library riscv-riscv and riscv-fdt

hyperion009, 2022-11-15:
Hi 
I am trying to run a test case under core_ibex "make TEST=riscv_machine_mode_rand_test ITERATIONS=1" and get an issue:  
**subprocess.CalledProcessError: Command'['pkg-config', '--exists',  'riscv-riscv', 'riscv-disasm', 'risccv-fdt']' return non-zero exit status 1**
I compiled the source codes of "riscv-isa-sim-master" from https://github.com/lowRISC/riscv-isa-sim/blob/master/
and found there is only  **riscv-disasm.pc and riscv-fesvr.pc** in there.   
Did I miss something?  or I need to create pc file for riscv-riscv and riscv-fdt manually?

GregAC, 2022-11-15:
You need to build the `ibex-cosim` branch: https://github.com/lowRISC/riscv-isa-sim/tree/ibex_cosim

hyperion009, 2022-11-16:
thanks, Greg

closed 2022-11-16 by GregAC