lowRISC/ibex #1940: Missing a module tc_clk_gating

hyperion009, 2022-11-16:
Hi 
when the flow was compiling the RTL files, the following issue popped up:  
[image]  
It seems that there is no module tc_clk_gating after grepping the whole tree.  my compilied version is  ibex-pulpissimo-v6.1. 
[image]

DerakhshanAli, 2023-11-01:
hi.
I am also facing this problem. (  fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast )
Is there still no solution?
[image]

JackySu, 2025-05-23:
you can copy paste this to anywhere in your RTL and it will work
```systemverilog
// Description: Behavioral model of an integrated clock-gating cell (ICG)
module tc_clk_gating #(
  /// This paramaeter is a hint for tool/technology specific mappings of this
  /// tech_cell. It indicates wether this particular clk gate instance is
  /// required for functional correctness or just instantiated for power
  /// savings. If IS_FUNCTIONAL == 0, technology specific mappings might
  /// replace this cell with a feedthrough connection without any gating.
  parameter bit IS_FUNCTIONAL = 1'b1
)(
   input  logic clk_i,
   input  logic en_i,
   input  logic test_en_i,
   output logic clk_o
);

  logic clk_en;

  always_latch begin
    if (clk_i == 1'b0) clk_en <= en_i | test_en_i;
  end

  assign clk_o = clk_i & clk_en;

endmodule
```

reference: https://github.com/pulp-platform/tech_cells_generic/blob/7968dd6e6180df2c644636bc6d2908a49f2190cf/src/rtl/tc_clk.sv#L31