lowRISC/ibex #1955: Error when I make in ibex UVM DV

Zhengshuhang, 2022-11-22:
Hi, I am a new learner in ibex DV,I tried IBEX DV in vcs2016,questa2021,xcelium2018 but failed. I guess it's because my SV simulator version,so I changed to CentOS7.9,python 3.8.10,VCS2020.03,but there are still problems as follow:
[image]
the corresponding compile_tb.log as following:
[compile_tb_vcs2020_ibex.log](https://github.com/lowRISC/ibex/files/10065680/compile_tb_vcs2020_ibex.log)
I find the following error in log: 
[image]

Can you help me?

GregAC, 2022-11-25:
This may be a genuine issue with our DV code, just one that only causes issues under VCS. So I will leave this open as a reminder for us to look into when we have time to do so.

Zhengshuhang, 2022-11-27:
Thank you for your kind reply. I will patiently wait for you to help me solve this problem. But I have a small question. I specifically use the VCS2020.3 mentioned in ibex, which is currently in use. Is it only my machine that has this problem？

H011YM00N, 2023-03-04:
Hi Sirs, 
I also encountered this compile issues using VCS/Q-2020.03-SP2-7. 
And I solved this issue by modifying the file "ibex_mem_intf_agent.pkg.sv" to append `export ibex_mem_intf_pkg::*;` behind `import ibex_mem_intf_pkg::*;`

IEEE_Std1800-2017, Section 26.6 Exporting imported names from packages.
> By default, declarations imported into a package are not visible by way of subsequent imports of that package. Package export declarations allow a package to specify that imported declarations are to be made visible in subsequent imports. A package export may precede a corresponding package import.

Hope this can help you. 
Thanks