lowRISC/ibex #1956: .S file is not generated for riscv_arithmetic_basic_test

Shaileshkavar, 2022-11-25:
Hello,

When I am trying to run UVM tests, using the below command to generate riscv_arithmetic_basic_test for example:

make instr_gen_run SIMULATOR=questa TEST=riscv_arithmetic_basic_test ISS=spike ISA=rv32imc SEED=1 ISA=rv32imc

Error:

placeholder_dir /tmp/tmps1r1tvid/@@Placeholder@@
build_dir /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/build/instr_gen
placeholder_dir /tmp/tmps1r1tvid/@@Placeholder@@
build_dir /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/build/instr_gen
file_copies [('gen.log', 'gen.log', True), ('test_0.S', 'test.S', False)]
src_path: /tmp/tmps1r1tvid/gen.log
dst_path: /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/gen.log
Compiling generated test assembly to create binary at out/run/tests/riscv_arithmetic_basic_test.1/test.bin
cc1: fatal error: /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.1/test.S: No such file or directory
compilation terminated.

And when I tried to generate the assembly file for the same test in google-riscv, it is generated properly.

[shailesh.kavar@vnc12 ~/riscv-ibex/ibex/vendor/google_riscv-dv]% python3 run.py --test riscv_arithmetic_basic_test --simulator questa Fri, 25 Nov 2022 11:18:49 INFO Creating output directory: out_2022-11-25
Fri, 25 Nov 2022 11:18:49 INFO Processing regression test list : /users3/shailesh.kavar/riscv-ibex/ibex/vendor/google_riscv-dv/target/rv32imc/testlist.yaml, test: riscv_arithmetic_basic_test
Fri, 25 Nov 2022 11:18:49 INFO Processing regression test list : /users3/shailesh.kavar/riscv-ibex/ibex/vendor/google_riscv-dv/yaml/base_testlist.yaml, test: riscv_arithmetic_basic_test
Fri, 25 Nov 2022 11:18:49 INFO Found matched tests: riscv_arithmetic_basic_test, iterations:2
Fri, 25 Nov 2022 11:18:49 INFO Processing simulator setup file : /users3/shailesh.kavar/riscv-ibex/ibex/vendor/google_riscv-dv/yaml/simulator.yaml
Fri, 25 Nov 2022 11:18:49 INFO Found matching simulator: questa
Fri, 25 Nov 2022 11:18:49 INFO Building RISC-V instruction generator
Fri, 25 Nov 2022 11:19:04 INFO Running RISC-V instruction generator
Fri, 25 Nov 2022 11:19:04 INFO Generating 2 riscv_arithmetic_basic_test
Fri, 25 Nov 2022 11:19:04 INFO Running riscv_arithmetic_basic_test with 1 batches
Fri, 25 Nov 2022 11:19:04 INFO Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Fri, 25 Nov 2022 11:19:16 INFO Compiling out_2022-11-25/asm_test/riscv_arithmetic_basic_test_0.S
Fri, 25 Nov 2022 11:19:16 INFO Converting to out_2022-11-25/asm_test/riscv_arithmetic_basic_test_0.bin
Fri, 25 Nov 2022 11:19:16 INFO Compiling out_2022-11-25/asm_test/riscv_arithmetic_basic_test_1.S
Fri, 25 Nov 2022 11:19:16 INFO Converting to out_2022-11-25/asm_test/riscv_arithmetic_basic_test_1.bin
Fri, 25 Nov 2022 11:19:16 INFO Processing ISS setup file : /users3/shailesh.kavar/riscv-ibex/ibex/vendor/google_riscv-dv/yaml/iss.yaml

GregAC, 2022-11-25:
I note you're using Questa. This isn't a simulator we have access to so we're unable to reproduce it. I suspect no-one's tried to run the testbench with Questa for a while and the flow has had many changes recently. So it's likely broken.

Shaileshkavar, 2022-11-25:
Thanks GregAC.

It would be useful if you could post the most recent testbench status for questasim since this problem is with assembly but I had to modify a few files to compile the testbench.