lowRISC/ibex #1958: [Questasim] Simulation error:  Failed to find design unit

Shaileshkavar, 2022-11-29:
Hello,

With Questasim, I'm having the following simulation problem. Does anyone know how to fix this? 

Error:
`# vsim -c -do "run -a; quit -f" "+designfile" -f /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/build/tb/top.list "+signature_addr=8ffffffc" "+test_timeout_s=1800" "+disable_cosim=1" -sv_seed 1 -suppress 8323 2247 "+UVM_TESTNAME=core_ibex_csr_test" "+UVM_VERBOSITY=UVM_LOW" "+bin=/users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/run/tests/riscv_csr_test.1/test.bin" "+ibex_tracer_file_base=/users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/run/tests/riscv_csr_test.1/trace_core" -l /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/run/tests/riscv_csr_test.1/sim.log -g/core_ibex_tb_top/PMPNumRegions=4 -g/core_ibex_tb_top/ICache=0 -g/core_ibex_tb_top/BranchTargetALU=0 -g/core_ibex_tb_top/SecureIbex=0 -g/core_ibex_tb_top/PMPGranularity=0 -g/core_ibex_tb_top/WritebackStage=0 -g/core_ibex_tb_top/ICacheECC=0 -g/core_ibex_tb_top/MHPMCounterWidth=40 -g/core_ibex_tb_top/MHPMCounterNum=0 -g/core_ibex_tb_top/BranchPredictor=0 -g/core_ibex_tb_top/DbgTriggerEn=0 -g/core_ibex_tb_top/PMPEnable=0 -g/core_ibex_tb_top/ICacheScramble=0 -g/core_ibex_tb_top/RV32E=0 
# Start time: 17:14:31 on Nov 29,2022
# Loading /tmp/shailesh.kavar@vnc12.einfochips.com_dpi_14217/linux_x86_64_gcc-5.3.0/export_tramp.so
# ** Note: (vsim-3812) Design is being optimized...
# ** Error: (vopt-13130) Failed to find design unit 2247.
#         Searched libraries:
#             work
# Optimization failed
# Error loading design
# End time: 17:14:34 on Nov 29,2022, Elapsed time: 0:00:03
# Errors: 1, Warnings: 0, Suppressed Errors: 1`

Run Command:
make ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1 SIMULATOR=questa IBEX_CONFIG=small

--
BR,
Shailesh

NikosDelijohn, 2022-12-12:
Where is your `vlog/vopt` section ? Also which version of `questa` sim is this?

Shaileshkavar, 2022-12-23:
The issue is not present with the most recent releases. Thanks!

closed 2022-12-23 by Shaileshkavar