lowRISC/ibex #1961: Ask a question related to Ibex

NikosDelijohn, 2022-12-06:
## My Environment

I wish to create a logic simulation environment while using ibex as a core. I have a question related on the structure of the core though. 

1.  Besides manual modification of the HDL, is there a way of including the Register File (FF) in the Instruction Decode stage ? In the current version I see that the RF is instantiated in the top module rather than in the ID whereas I would like to embed it in the core.
2.  Is there a pre-written TB for the core in SystemVerilog ? I can't seem to find  something relevant. Do i have to create a TB from scratch?
3. Is there a .tcl synthesis script that can be used instead of the yosys synthesis one?

**EDA tool and version:**

QuestaSIM 10.5c_4
Design Compiler Version S-2021.06-SP4 (for synthesis)

**Operating system:**

OpenSUSE Leap 42.3 (EOL)

**Version of the Ibex source code:**

Latest

NikosDelijohn, 2022-12-12:
> > Besides manual modification of the HDL, is there a way of including the Register File (FF) in the Instruction Decode stage ? In the current version I see that the RF is instantiated in the top module rather than in the ID whereas I would like to embed it in the core.
> 
> There isn't. It's done this way as it's required to support our dual core lockstep functionality, so you'd have to move it manually. Do you have a particular motivation for doing so? Synthesis tools shouldn't have any issues with optimising regardless of where it lives in the hierarchy though if you're using a strictly hierarchical synthesis flow it could impact results.
> 
> > Is there a pre-written TB for the core in SystemVerilog ? I can't seem to find something relevant. Do i have to create a TB from scratch?
> 
> Yes we have an extensive UVM based verification environment: https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex with documentation on the environment and testbench itself: https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html the coverage plan and test plan: https://ibex-core.readthedocs.io/en/latest/03_reference/coverage_plan.html and https://ibex-core.readthedocs.io/en/latest/03_reference/testplan.html and the co-simulation setup we use for checking: https://ibex-core.readthedocs.io/en/latest/03_reference/cosim.html
> 
> It should be usable under Questa but may not work out of the box as we do not run Questa internally to lowRISC
> 
> > Is there a .tcl synthesis script that can be used instead of the yosys synthesis one?
> 
> Unfortunately not, the yosys flow is the only one we have developed so far.

Thank you for your reply.

I managed to make some changes and incorporate the use of the following EDA flows in the core:

- Synthesis using Design Compiler.
- Logic Simulation via QuestaSIM on RTL/Gate-Level with and without GUI.
- Fault Simulation of (E)VCD files on the `ibex_top` using Z01X for Stuck-At faults. 

Would these be of interest to you ? If so I could submit a pull request so that you can check. 

Regards
-Nick

sadjadtu, 2023-09-04:
> Thank you for your reply.
>
> I managed to make some changes and incorporate the use of the following EDA flows in the core:
>
> * Synthesis using Design Compiler.
> * Logic Simulation via QuestaSIM on [...]
Hi Nik,
I want to verify my designed system using Stuck-At faults which affects all the signals automatically. Can you help me how to do that?