lowRISC/ibex #1965: memory initialization

sadjadtu, 2022-12-22:
My program uses some of the memory addresses as primary inputs. Is there any way to initialize these addresses before stating simulation? I am trying to simulate my software on the Ibex Simple System.

GregAC, 2023-01-19:
You can initialise those addresses by giving them values in the .elf you load into the verilator simulation. Precise details of how to do this depend upon your toolchain.

Alternatively you could look at the memory initialisation code (https://github.com/lowRISC/ibex/blob/master/vendor/lowrisc_ip/dv/verilator/cpp/verilator_memutil.cc) and extend it to initialise specification addresses to your needs.

sadjadtu, 2023-01-19:
I am using RISC-V Compiler Toolchain, GCC version 10.2.0